gyroflow/gyroflow #56: [Bug]: Unexpected offset in blackbox data (around 16 seconds)

kcsajtai, 2022-02-04:
### Gyroflow version

1.0

### What operating system are you using?

Windows 10

### What GPU are you using?

Nvidia GTX1050

### What happened?

The blackbox data offset is around 16 seconds instead of the realistic sub 100ms 

Log file: https://drive.google.com/file/d/1U_gbSTmRwTgzxNju1N2Hr0IF7TIsLgCQ/view?usp=sharing

Video: https://drive.google.com/file/d/1gv2aEWBkTfs6eD3ytMmlptbBHzeonzJj/view?usp=sharing

kcsajtai, 2022-02-04:
Made some tests, and different files give different offsets, and its in direct relation with the time value in the first row of the logs. The file above has a time value of 16422088 and the offset is cca 16 seconds. An other file has a time value of 60320266 and the offset is around 60 seconds, and so on.

AdrianEddy, 2022-02-04:
ok, it'll be easy fix then :)

RD5500, 2022-02-05:
My solution is too add 16 seconds to the rough gyro offset. For example if the actual offset is 67 seconds, add 16 to make it 83. Do this from the beginning, if you correct it after you perform an auto sync using the actual offset, it won't change a thing in my experience. Explained in more detail here in my vid:
https://youtu.be/_YvfFJTC3Bo
Overall, this update is amazing. syncing is much easier and rendering is way faster (at least for me). It's understandable that this v1.0 RC1 has a bug like this and i'm happy to report more if there are any :)

kcsajtai, 2022-02-05:
> My solution is too add 16 seconds to the rough gyro offset. For example if the actual offset is 67 seconds, add 16 to make it 83. Do this from the beginning, if you correct it after you perform an [...]

This is not a solution, this is a workaround. Since there were efforts being made to make the initial offset zero (which already worked in 0.3) there is really no need to introduce a random variable into to the workflow. And syncing has issues as well :)  All other points are valid, amazing, fast and such. But a bug is a bug :)

AdrianEddy, 2022-02-07:
Fixed in 785ffa3f2c715accca113324f1819bacc40459c7

closed 2022-02-07 by AdrianEddy