gyroflow/gyroflow #57: [Feature request]: Flatpak build for Linux

jimmac, 2022-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Description

While Gyroflow is available as an AppImage, the statement "`.AppImage` should work everywhere" is sadly a wish:

```
$ ./Gyroflow-linux64.AppImage 
/tmp/.mount_GyroflfH8T7x//gyroflow: error while loading shared libraries: libnsl.so.1: cannot open shared object file: No such file or directory
```

By releasing it as a Flatpak, it'd cover all Linux distributions and would be the easiest option to install the app for Linux users.

Flatpak's can be hosted on Flathub servers: https://flathub.org/

froqstar, 2022-03-25:
Being a total flatpak noob, I tried to create a flatpak manifest. 
I currently struggle with getting the opencv dependency to work.
It needs `OPENCV_LINK_PATHS` environment variable needs to be set, not sure whether compiling opencv as a dependency/module is the right way or actually needed, as it doesn't set the variable of course :grimacing: 

In case anybody wants to pick this up, here is my current state:
```yaml
app-id: xyz.gyroflow.Gyroflow
runtime: org.kde.Platform
runtime-version: '5.15-21.08'
sdk: org.kde.Sdk
sdk-extensions:
  - "org.freedesktop.Sdk.Extension.rust-stable"
  - "org.freedesktop.Sdk.Extension.llvm13"
command: gyroflow

build-options:
  env: 
    CARGO_HOME: /run/build/gyroflow/cargo
    
  append-path: /usr/lib/sdk/rust-stable/bin
  prepend-path: /usr/lib/sdk/llvm13/bin
  prepend-ld-library-path: /usr/lib/sdk/llvm13/lib
  build-args:
    - --share=network
    
modules:
  - name: opencv
    buildsystem: cmake
    builddir: true
    sources:
      - type: archive
        url: https://github.com/opencv/opencv/archive/4.5.5.zip
        sha256: fb16b734db3a28e5119d513bd7c61ef417edf3756165dc6259519bb9d23d04e2
  - name: gyroflow
    buildsystem: simple
    build-commands: 
      - cargo build --release
    sources: 
      - type: git
        url: https://github.com/gyroflow/gyroflow
        
finish-args:
  - --filesystem=home:rw 
  - --socket=wayland
  - --share=ipc
  - --socket=fallback-x11
```

razzeee, 2022-10-12:
I seem stuck for now https://github.com/razzeee/xyz.gyroflow.Gyroflow

```
     Running `rustc --crate-name qttypes --edition=2018 /run/build/gyroflow/cargo/vendor/qttypes/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts,future-incompat --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C embed-bitcode=no --cfg 'feature="default"' --cfg 'feature="qtquick"' --cfg 'feature="qtquickcontrols2"' --cfg 'feature="qtwidgets"' --cfg 'feature="required"' -C metadata=c129dad2d1c81fd9 -C extra-filename=-c129dad2d1c81fd9 --out-dir /run/build/gyroflow/target/release/deps -L dependency=/run/build/gyroflow/target/release/deps --extern cpp=/run/build/gyroflow/target/release/deps/libcpp-a6560aeb0c6ad598.rmeta --cap-lints allow -C 'link-args=-Wl,-rpath,$ORIGIN,-rpath,$ORIGIN/lib' -L native=/run/build/gyroflow/target/release/build/qttypes-39b31af0b7294758/out -L /usr/lib/x86_64-linux-gnu -l static=rust_cpp_generated -l stdc++ -l Qt5Core -l Qt5Gui -l Qt5Widgets -l Qt5Quick -l Qt5Qml -l Qt5QuickControls2`
     Running `/run/build/gyroflow/target/release/build/qmetaobject-5fea7fb6594f1d9a/build-script-build`
     Running `/run/build/gyroflow/target/release/build/gyroflow-d119bf00ff683abf/build-script-build`
error: failed to run custom build command for `gyroflow v1.2.0 (/run/build/gyroflow)`

Caused by:
  process didn't exit successfully: `/run/build/gyroflow/target/release/build/gyroflow-d119bf00ff683abf/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: NotPresent', build.rs:143:87
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
Error: module gyroflow: Child process exited with code 101
```

I was seeing a different problem on a different system, but not sure why.

nbenitez, 2022-10-12:
> note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

Doing that would help to see the cause of the crash.

AdrianEddy, 2022-10-12:
I didn't try it, but I don't see Qt 6.4.0 in your .json? `qmake` needs to be in PATH. Qt 5 doesn't work

razzeee, 2022-10-12:
Right, forgot about qt6, as I previously had problems with that and side stepped it, 

Anyway changed that now, running into a follow up problem.

razzeee, 2022-10-13:
This is the current one, which is puzzling to me

```
     Running `/run/build/gyroflow/target/release/build/gyroflow-28bf6273379e2339/build-script-build`
error: failed to run custom build command for `gyroflow v1.2.0 (/run/build/gyroflow)`

Caused by:
  process didn't exit successfully: `/run/build/gyroflow/target/release/build/gyroflow-28bf6273379e2339/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: NotPresent', build.rs:143:87
  stack backtrace:
     0: rust_begin_unwind
               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library/std/src/panicking.rs:584:5
     1: core::panicking::panic_fmt
               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library/core/src/panicking.rs:142:14
     2: core::result::unwrap_failed
               at /rustc/4b91a6ea7258a947e59c6522cd5898e7c0a6a88f/library/core/src/result.rs:1805:5
     3: core::result::Result<T,E>::unwrap
     4: build_script_build::main
     5: core::ops::function::FnOnce::call_once
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
warning: build failed, waiting for other jobs to finish...
Error: module gyroflow: Child process exited with code 101
```

AdrianEddy, 2022-10-13:
hmm, weird, there's no `unwrap()` on line 143 in build.rs, are you building latest master?

razzeee, 2022-10-13:
No, that's supposed to be the 1.2.0 tag

marbub, 2024-03-20:
The AppImage is working for me but flatpak is much more comfortable. I would like to help but I have no clue about the Rust build system. I built already a flatpak but that was with CMake.

siefkenj, 2024-04-24:
I have made a little progress on this, trying to run the pre-compiled version but wrapped in a flatpak container. However, I cannot find a flatpak runtime that comes with `libc++.so`. Is it possible to statically link `libc++.so`? Or does someone know of a flatpak runtime that already includes it?

hfiguiere, 2024-04-24:
> I have made a little progress on this, trying to run the pre-compiled version

That's the problem. build from source. your binary is definitely not portable.

siefkenj, 2024-04-25:
> > I have made a little progress on this, trying to run the pre-compiled version
> 
> That's the problem. build from source. your binary is definitely not portable.

Gyroflow cannot be built without libcxx, as far as I know.

siefkenj, 2024-04-28:
I got it working!! At least on my machine. The current approach uses the pre-built linux binaries and builds and installs the missing `libc++.so` and `libkrb5`.

If others can reproduce, I may open a PR. Also, I know very little about flatpak, so I am sure this manifest can be improved. It also doesn't contain information needed for a `.desktop` file...

In any case, here's what to do to try and reproduce:
 1. Make a `flatpak` directory (this will be the working directory for the build)
 2. Follow the setup [here](https://docs.flatpak.org/en/latest/first-build.html) to install `flatpak-builder`
 3. `flatpak install org.kde.Platform org.kde.Sdk org.freedesktop.Sdk.Extension.rust-stable org.freedesktop.Sdk.Extension.llvm17`
 4. Create a `xyz.gyroflow.Gyroflow.yml` file with the following contents:
```yml
id: xyz.gyroflow.Gyroflow
runtime: org.kde.Platform
runtime-version: '6.7'
sdk: org.kde.Sdk
sdk-extensions:
  - "org.freedesktop.Sdk.Extension.rust-stable"
  - "org.freedesktop.Sdk.Extension.llvm17"
command: gyroflow

build-options:
  env: 
    CARGO_HOME: /run/build/gyroflow/cargo
    
  append-path: /usr/lib/sdk/rust-stable/bin
  append-path: /run/build/gyroflow/cargo/bin
  prepend-path: /usr/lib/sdk/llvm17/bin
  prepend-ld-library-path: /usr/lib/sdk/llvm17/lib

modules:
  - name: libcxx
    buildsystem: simple
    builddir: true
    sources: 
      - type: archive
        url: https://github.com/llvm/llvm-project/releases/download/llvmorg-17.0.6/llvm-project-17.0.6.src.tar.xz
        sha256: 58a8818c60e6627064f312dbf46c02d9949956558340938b71cf731ad8bc0813
    build-options:
      env:
        CMAKE_INSTALL_PREFIX: /app
    build-commands:
      - echo "Building libcxx"
      - mkdir build
      - cmake -G Ninja -S runtimes -B build -DLLVM_ENABLE_RUNTIMES="libcxx;libcxxabi"
      - ninja -C build cxx cxxabi
      - ninja -C build install-cxx install-cxxabi

  - name: krb5
    buildsystem: simple
    builddir: true
    sources:
      - type: archive
        url: https://web.mit.edu/kerberos/dist/krb5/1.21/krb5-1.21.2.tar.gz
        sha256: 9560941a9d843c0243a71b17a7ac6fe31c7cebb5bce3983db79e52ae7e850491
    build-commands:
      - echo "Building krb5"
      - ls
      - cd src && ./configure --prefix=/app
      - cd src && make -j8
      - cd src && make install

  - name: gyroflow
    buildsystem: simple
    build-commands:
      - pwd
      - ls
      # This is surely not the correct thing to do...
      - cp -r * /app/bin/
    sources: 
      - type: archive
        url: https://github.com/gyroflow/gyroflow/releases/download/v1.5.4/Gyroflow-linux64.tar.gz
        sha256: 51ab98bfa3ff28dfe75fb75425e14f7ac2077957b31b703d232a07cc4318635c
        
finish-args:
  - --filesystem=home:rw 
  - --socket=wayland
  - --share=ipc
  - --socket=fallback-x11
  - --device=dri
  - --share=network
  - --socket=pulseaudio
```
 5. Make a `build-dir` directory
 6. Run `flatpak-builder --force-clean build-dir xyz.gyroflow.Gyroflow.yml`
 7. If that worked, you can locally install the flatpak with `flatpak-builder --user --install --force-clean build-dir xyz.gyroflow.Gyroflow.yml`
 8. Then try to run it with `flatpak run xyz.gyroflow.Gyroflow`

razzeee, 2024-04-28:
The krb5 part didn't work for me, so I changed it based on https://github.com/flathub/com.freerdp.FreeRDP/blob/20203265155edc33a3324bada4047ff02af1b28c/modules/krb5.json#L2

```
id: xyz.gyroflow.Gyroflow
runtime: org.kde.Platform
runtime-version: '6.7'
sdk: org.kde.Sdk
sdk-extensions:
  - "org.freedesktop.Sdk.Extension.rust-stable"
  - "org.freedesktop.Sdk.Extension.llvm17"
command: gyroflow

build-options:
  env:
    CARGO_HOME: /run/build/gyroflow/cargo

  append-path: /usr/lib/sdk/rust-stable/bin
  append-path: /run/build/gyroflow/cargo/bin
  prepend-path: /usr/lib/sdk/llvm17/bin
  prepend-ld-library-path: /usr/lib/sdk/llvm17/lib

modules:
  - name: libcxx
    buildsystem: simple
    builddir: true
    sources:
      - type: archive
        url: https://github.com/llvm/llvm-project/releases/download/llvmorg-17.0.6/llvm-project-17.0.6.src.tar.xz
        sha256: 58a8818c60e6627064f312dbf46c02d9949956558340938b71cf731ad8bc0813
    build-options:
      env:
        CMAKE_INSTALL_PREFIX: /app
    build-commands:
      - echo "Building libcxx"
      - mkdir build
      - cmake -G Ninja -S runtimes -B build -DLLVM_ENABLE_RUNTIMES="libcxx;libcxxabi"
      - ninja -C build cxx cxxabi
      - ninja -C build install-cxx install-cxxabi

  - name: krb5
    buildsystem: autotools
    subdir: src
    sources:
      - type: archive
        url: https://kerberos.org/dist/krb5/1.21/krb5-1.21.2.tar.gz
        sha256: 9560941a9d843c0243a71b17a7ac6fe31c7cebb5bce3983db79e52ae7e850491
        x-checker-data:
          - type: anitya
          - project-id: 13287
          - stable-only: true
          - url-template: https://kerberos.org/dist/krb5/${version0}.${version1}/krb5-${version0}.${version1}.${version2}.tar.gz

  - name: gyroflow
    buildsystem: simple
    build-commands:
      - pwd
      - ls
      # This is surely not the correct thing to do...
      - cp -r * /app/bin/
    sources:
      - type: archive
        url: https://github.com/gyroflow/gyroflow/releases/download/v1.5.4/Gyroflow-linux64.tar.gz
        sha256: 1b2cccb33f2f3f00a8e22daf77d51407552905604b3e3d000220fd80581b19d0

finish-args:
  - --filesystem=home:rw
  - --socket=wayland
  - --share=ipc
  - --socket=fallback-x11
  - --device=dri
  - --share=network
  - --socket=pulseaudio
```

I could build and start it fine after these changes

In the linked repo above, you can find a starting point for the appdata and all other missing files, feel free to use them as a starting point.