gyroflow/gyroflow #58: [Bug]: Entering LPF on Motion Data Crashes the app

deadbytefpv, 2022-02-04:
### Gyroflow version

1.0.0-RC1

### What operating system are you using?

Windows 10

### What GPU are you using?

GTX 1070Ti

### What happened?

When I try entering a value for the Low Pass Filter for the Motion Data, the app closes by itself. No error messages.

AdrianEddy, 2022-02-04:
Can you run "Gyroflow_with_console.bat" and paste the output after it crashes?

deadbytefpv, 2022-02-04:
I think I found the issue. I'll post back should it occur again.

Thank you for the fast response!

deadbytefpv, 2022-02-04:
Closing the issue.. It was a user error. 

:)

closed 2022-02-04 by deadbytefpv

AdrianEddy, 2022-02-04:
Even in case of user error, it should never crash but show an error message in the UI. If you have a steps to reproduce the crash, please share, it'll help to improve the app :)