gyroflow/gyroflow #59: [Bug]: Image orientation wrong on initial load

sweebee, 2022-02-04:
### Gyroflow version

v1.0.0-rc1

### What operating system are you using?

MacOS 12.1

### What GPU are you using?

AMD Radeon Pro 5300M

### What happened?

When opening an Insta360 go2 file the orientation is set to 180 degrees, which seems good. But the image is upside down. When i set the orientation to 0 and then back to 180 it looks good.

[image]

sweebee, 2022-02-04:
made a video of the orientation bug on load: https://www.youtube.com/watch?v=QcMTATE5AO0 in the beginning i have to set the orientation to 0 and then back to 180.

ElvinC, 2022-02-07:
Fixed with https://github.com/gyroflow/gyroflow/pull/109

closed 2022-02-07 by ElvinC