gyroflow/gyroflow #60: [Bug]: toggling stabilisation also changes orientation

sweebee, 2022-02-04:
### Gyroflow version

v1.0.0-rc1

### What operating system are you using?

MacOS 12.1

### What GPU are you using?

AMD Radeon Pro 5300M

### What happened?

When toggling stabilisation on/off the orientation also changes, seems like a bug to me. might be related to #59 

[image]

alexagv, 2022-02-04:
@sweebee thanks for the report, could you by any chance share a short Go2 clip so we can reproduce the issue?
Send it via WeTransfer etc? 

And how was the camera mounted in this clip? Was it sideways or upside down by any chance?

sweebee, 2022-02-04:
the file: https://we.tl/t-XFx9w7l9W5

It was shot handheld with the lens on top (vertical) like this:

[image]

alexagv, 2022-02-04:
@sweebee can you send me a screenshot of the entire application so I can see all settings, lens profile etc?
Can't reproduce it here (also on Mac)

alexagv, 2022-02-04:
@sweebee aah, now I see. Managed to reproduce it. 
Seems like the Rotation element only affects the "stabilized" preview and not the original preview 👍🏻

sweebee, 2022-02-04:
great. still made a video: https://www.youtube.com/watch?v=QcMTATE5AO0

mycosd, 2022-06-03:
I don't see this as a bug, it's just the label of the button is misleading. It should be called something like "toggle processing". As it just toggles between the original video and the processed video (not just stabilization).

AdrianEddy, 2022-06-07:
but the main point of that button is to compare original with stabilized footage and if it rotates after toggling you lose that ability to compare. Fixing this is simple, we can just rotate QML element using `rotation` property when the stab toggle is off

mycosd, 2022-06-07:
If you only want to compare the stabilization, you would have to apply the lens correction too.