gyroflow/gyroflow #61: [Bug]:

JackJ87, 2022-02-04:
### Gyroflow version

1.0.0

### What operating system are you using?

Windows 10

### What GPU are you using?

AMD Ryzen 9 3950x

### What happened?

I've loaded a video file into gyro flow, chosen the lens profile. When I then chose my blackbox file, the application crashes straight away with no messages.

AdrianEddy, 2022-02-04:
Please upload that blackbox file

JackJ87, 2022-02-04:
[LOG00009.CSV](https://github.com/gyroflow/gyroflow/files/8004952/LOG00009.CSV)

Here is the logfile.

AdrianEddy, 2022-02-04:
thank you, it should be fixed in a few days

JackJ87, 2022-02-04:
Amazing. Thank you! 
Absolutely game changing things you guys are doing!

Fersy2003, 2022-02-05:
here the same, win11 *.bbl export with blackbox explorer, when loading => crash to desktop
[BTFL_BLACKBOX_LOG_Titan_DC5_20220205_174712.BBL_video2.csv](https://github.com/gyroflow/gyroflow/files/8008492/BTFL_BLACKBOX_LOG_Titan_DC5_20220205_174712.BBL_video2.csv)
p

valtermendonca, 2022-02-05:
When I choose the ProRes file, GF assume that same file in the Motion's Data. When I try to change it to the BBL file, GF shuts down. By Adrian Eddy request im attaching the BBL file.
Another thing that happen, its my Computer keeps asking the authorisation from GF to access to the Desktop Folder, even when I gave GF access to the whole computer files.
I must say that this computer is brand new for me (MAC M1), and I did run GF 0.3.0 on my previous MAC with 10 years old with no issues at all with great results in Stabilisation.
If I am the only one with issue, maybe is same setting on my Computer...
Thanks for effort in this BIG Project
Hope that help
[loglagoalut3.BBL.zip](https://github.com/gyroflow/gyroflow/files/8008833/loglagoalut3.BBL.zip)

ilyafpv, 2022-02-07:
This one crashes on open as well
https://we.tl/t-9kxRABsSym

ilya-epifanov, 2022-02-08:
The issue has been fixed, no new reports are needed for now. Please wait for an updated build

AdrianEddy, 2022-02-08:
Fixed in e532d34025823dacbeaf2e8fb90708b3df56a90c
Download here: https://github.com/gyroflow/gyroflow/suites/5220852577/artifacts/160018759

closed 2022-02-08 by AdrianEddy