gyroflow/gyroflow #62: [Bug]: Color shift in ProRes export of the stabilized footage

leitmanr, 2022-02-04:
### Gyroflow version

1.0

### What operating system are you using?

Windows 10 Pro 21H2 19044.1466

### What GPU are you using?

EVGA RTX 3080ti FTW3

### What happened?

There seems to be a slight color shift issue with the prores encoder of the GF.
I have exported a EBU chart from resolve and run it through GF
Then I have imported the original file and the"stabilized" one above it and isolated the colors and compared various exports from GF with the original file.
[image]

here is a pic of how the isolated regions of the "stabilized" footage looked 
[image]

and here are some comparisons h.264/265 gpu or cpu encoded looked exactly the same but with prores exports there is a slight color/value shift in white, yellow, cyan, green and magenta bars. PNG's are way off but I am really not worried about them (I assume it's a colorspace tag issue with the PNG's)
here are the tests of both Prores and png's
Prores 4444
[image]

Prores 4444XQ 1000Mbps
[image]

PNG8bit
[image]

PNG16bit
[image]

AdrianEddy, 2022-02-04:
Thanks for testing, we'll look into it. Probably an issue with setting correct color parameters for the encoder. PNG also needs to be fixed, so that's for that comparison too

AdrianEddy, 2022-02-08:
Was the input file 4:2:2 10-bit prores? The exactly the same pixel format as the export?

leitmanr, 2022-02-08:
After downloading a build with fixed prores export speed I've did the color shift test again and the slight color/value shift in white, yellow, cyan, magenta, red and blue colors is still present.. it is minor (from 1 to 2 value change in channels) , but it's still there...

[image]

The output values of RGB channels change in
white from 255/255/255 to 254/254/254
yellow from 192/191/-1 to 191/191/-1
cyan from -1/192/191 to -1/191/190
magenta from 192/0/193 to 191/0/192
red from 192/0/0 to 191/0/0
blue from -1/0/192 to -1/0/190

this might be a negligible difference for 99,9% of users but when those colors combine and the footage is graded, it can grow and result in slight and noticeable difference in color of the output.

AdrianEddy, 2022-02-08:
Can you do a test with 4:2:2 10-bit input, and another one 4:2:0 10-bit input? There is a difference in implementation for these two and I want to see if it's any different

leitmanr, 2022-02-08:
the input file was h.265 (I am on windows, therefore I can't export prores, I'll try exporting DNxHR and test it with that...

AdrianEddy, 2022-02-08:
try 4:2:2 10-bit h265 as well

leitmanr, 2022-02-08:
well just ran into weird issue, GF would not open DNxHR HQX 10bit 422 file. shall I open it as a new issue?
[image]

from quick test with h265 10bit 420 -> proresHQ the color shift is only in the yellow color 
it goes from 192/191/0 to 191/191/0
[image][image]

[image]

leitmanr, 2022-02-08:
same behaviour on the yellow in the h265 10bit 422
[image]

AdrianEddy, 2022-02-08:
`DNxHR HQX 10bit 422 file. shall I open it as a new issue?`
Yes and upload that file

AdrianEddy, 2022-02-08:
ok one more test, can you reencode this file just using ffmpeg? with `-c:v prores_ks -pix_fmt yuv42210ple -profile hq`

AdrianEddy, 2022-02-08:
Hmm if this still looks the same in direct ffmpeg transcode then I'm afraid that's out of our control

leitmanr, 2022-02-08:
yup, I thought so.. we'll just have to learn to live with it for now I guess.. at least we know about it...

AdrianEddy, 2022-02-14:
Please test with this version: https://github.com/gyroflow/gyroflow/suites/5296956914/artifacts/164310984

AdrianEddy, 2022-02-19:
I believe this issue can be closed, all color conversions should be correct now (except for that tiny shift that also happens with ffmpeg)
https://gyroflow.xyz/devbuild/?autodownload
@leitmanr please double check and confirm there's no significant difference

leitmanr, 2022-02-19:
Will check but I agree. There is just that one minor shift caused by ffmpeg.
H.264/5 is perfect without any shifts.

closed 2022-02-24 by AdrianEddy