gyroflow/gyroflow #64: [Feature request]: Auto loop replay sync points when zoomed in

b14ckyy, 2022-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Description

If we zoom in to a sync point it would be nice to have an auto loop around the sync point as a center and 1s before and after it. 
This would help to visualize how well the sync is matching even before checking the graphs. 

It is already possible with the Trim feature but doing it automatically would be much easier and quicker, the same way as it is possible in Reelsteady Go.

AdrianEddy, 2022-11-21:
Anyone still wants this feature? I feel like since we have the new rs-sync algorithm, the need for manual adjusting like that is much lower

b14ckyy, 2022-11-21:
The autosync works pretty well but sometimes, especially in lower light conditions, manual sync still helps IMHO. Or if it is blurry a bit from the shutter speed.

closed 2022-11-21 by AdrianEddy