gyroflow/gyroflow #65: [Feature request]: Test AMD graphic cards

AdrianEddy, 2022-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Description

While it should work, AMD graphic cards weren't tested due to lack of AMD hardware.
If someone can figure out what is needed then probably helpful links are:

https://github.com/Intel-Media-SDK/MediaSDK/releases
https://gpuopen.com/advanced-media-framework/
https://trac.ffmpeg.org/wiki/HWAccelIntro

Yamakaky, 2022-02-06:
I have an AMD RX 6700XT, is there test files somewhere that I can test?

wesgtr, 2022-02-06:
I tested in am Radeon rx570 an receiving error about cuida.dll is missing. However cuda is for Nvidia how can I help in testing?

AdrianEddy, 2022-02-07:
Please test this one and report back: https://github.com/gyroflow/gyroflow/suites/5205488855/artifacts/159064470

avsaase, 2022-02-07:
I have the same AMD Radeon RX570 graphics card. With the latest build I don't get the cuda error but now it crashes when rendering with GPU Encoding, both with h264 and h265. Here's the log output:
```
20:55:43 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\Alexander\\Downloads\\gyroflow\\Gyroflow-windows64\\camera_presets"
20:55:43 [INFO] Latest version: v1.0.0-rc1, current version: v1.0.0-rc1
20:55:43 [INFO] Loaded 1290 lens profiles in 29.826ms
20:55:43 [INFO] OpenCL Platform: AMD Accelerated Parallel Processing, Device: Advanced Micro Devices, Inc. Ellesmere
[src\rendering\mod.rs:41] gpu_type = AMD
20:55:53 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
20:55:53 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "thread.video" ""
20:55:53 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
20:55:53 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
20:55:53 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
20:55:53 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
20:55:53 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x1f3c8652e20) QSize(1030, 579)
20:55:53 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "decoder.video" "MFT"
20:55:53 [DEBUG] (13) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
20:55:53 [INFO] surface size: 1280x720
20:55:53 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
20:55:53 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x1f3cfa30850) QSize(1280, 720)
20:55:57 [DEBUG] (2) gyroflow_core::smoothing::velocity_dampened: Max velocity: 0.0018223423497606378
20:55:58 [DEBUG] (2) gyroflow_core::smoothing::velocity_dampened: Max velocity: 0.0018223423497606378
20:55:58 [DEBUG] (3) gyroflow_core::smoothing::velocity_dampened: Max velocity: 0.0018223423497606378
20:56:03 [DEBUG] (2) gyroflow_core::smoothing::velocity_dampened: Max velocity: 0.0017853954135137189
20:56:11 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_DXVA2 0
20:56:11 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_DXVA2 (dxva2) with format Some(AV_PIX_FMT_DXVA2_VLD)

20:56:13 [INFO] Setting offset at 94194.1004: 789.8700 (cost 264.4802)
20:56:19 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_DXVA2 0
20:56:19 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_DXVA2 (dxva2) with format Some(AV_PIX_FMT_DXVA2_VLD)

20:56:20 [INFO] Setting offset at 484750.9356: 1186.9400 (cost 125.2867)
20:56:51 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
20:56:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_DXVA2 0
20:56:51 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_DXVA2 (dxva2) with format Some(AV_PIX_FMT_DXVA2_VLD)

20:56:51 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: Some("dxva2")
20:56:51 [DEBUG] (2) gyroflow::rendering: Possible encoders with AMD: [("h264_amf", true), ("h264_mf", true), ("h264_qsv", true), ("libx264", false)]
20:56:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_D3D11VA 0, for: h264_amf
20:56:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_D3D11VA
20:56:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_D3D11VA
20:56:51 [DEBUG] (2) gyroflow::rendering: proc.video_codec: Some("h264_amf")
20:56:51 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 542042, render_frame_count: 16245
[aac @ 000001F3CF6D4B40] Too many bits 32768.000000 > 12288 per frame requested, clamping to max
20:56:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Get transfer formats from GPU, frame.is_null: false, hw_ctx.is_null: false, format: DXVA2_VLD
20:56:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: [AV_PIX_FMT_NV12], frame.is_empty: true
20:56:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Codec formats: [AV_PIX_FMT_NV12, AV_PIX_FMT_YUV420P, AV_PIX_FMT_D3D11, AV_PIX_FMT_DXVA2_VLD]
20:56:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_D3D11VA), encoder_pixel_format: Some(NV12)
20:56:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: NV12, color range: JPEG
20:56:51 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: target_format: AV_PIX_FMT_NV12
[AVHWFramesContext @ 000001F3D77C9900] Could not create the texture (80070057)
20:56:51 [DEBUG] (2) gyroflow::rendering: Failed to initialize frame context. Error code: -1313558101

20:56:51 [DEBUG] (2) gyroflow::rendering: Failed to create encoder HW context.

20:56:51 [DEBUG] (2) gyroflow::rendering: process_frame: 1, timestamp_us: 0
```

AdrianEddy, 2022-02-07:
working on it

AdrianEddy, 2022-02-08:
Everyone please test on this version: https://github.com/gyroflow/gyroflow/suites/5212263246/artifacts/159464511

lukegluke, 2022-02-08:
Have tested last version with internal graphics, Intel UHD 630, Intel Core i7-8850H, Win 10 Pro. Now GPU encoding works and uses GPU in resources monitor and export ~5 times faster: ~1.5min against ~7.5 min for 30sec cut of 2.7k video from GoPro 6). Thanks!

Notice little visiual bug. Process can exceed 100% in the end and show invalid remaining time:
[image]

AdrianEddy, 2022-02-08:
Multiple people tested with AMD and Intel already and it worked, so I'm closing this issue. If anyone has further issues with these graphic cards still on the latest dev build, feel free to provide error and log

closed 2022-02-08 by AdrianEddy