gyroflow/gyroflow #68: [Bug]: INAV Logs detected but no gyro data extracted

b14ckyy, 2022-02-04:
### Gyroflow version

v1.0.0-rc1

### What operating system are you using?

Windows 11 - 22000.438

### What GPU are you using?

Core i7 7700K

### What happened?

INAV Blackbox Logs (TXT format) are detected and INAV version is shown but no gyro traces appear and no sync possible. 
decoded csv files are not accepted at all. 
Test log: 
[LOG00044.TXT](https://github.com/gyroflow/gyroflow/files/8006350/LOG00044.TXT)

[image]

Nemeton, 2022-02-05:
Same on Mac mini M1

Fersy2003, 2022-02-05:
same on win11 with inav 3.0.1

[LOG00004.TXT](https://github.com/gyroflow/gyroflow/files/8008503/LOG00004.TXT)

lukdut, 2022-02-07:
The save issue with inav 4.0.0. I tried to convert my *.txt log file into *.csv using blackbox-tools-4.0.1_win32, I got 450+Mb of gyro data, but it doesn't work neither with the error "Unsupported file format". The save *.csv output file works just fine with gyroflow-0.3.0-beta-win64 btw.

AlexZaiets, 2022-02-07:
same on windows 10 - 21H2(19044.1469), Gyroflow v1.0.0-rc1, INAV - 4.0.0
[LOG00037.TXT](https://github.com/gyroflow/gyroflow/files/8018381/LOG00037.TXT)

Fersy2003, 2022-02-07:
nope, doesn't works with inav 3.0.1 files

[LOG00004.TXT](https://github.com/gyroflow/gyroflow/files/8018598/LOG00004.TXT)

b14ckyy, 2022-02-07:
> nope, doesn't works with inav 3.0.1 files
> 
> [LOG00004.TXT](https://github.com/gyroflow/gyroflow/files/8018598/LOG00004.TXT)

Use Blackbox tools to decode into a CSV file and it works: 
[image]

Fersy2003, 2022-02-07:
> Use Blackbox tools to decode into a CSV file and it works: !

> > nope, doesn't works with inav 3.0.1 files
> > [LOG00004.TXT](https://github.com/gyroflow/gyroflow/files/8018598/LOG00004.TXT)
> 
> Use Blackbox tools to decode into a CSV file and it works: [image]

you mean converted with the blackbox_decode.exe?

b14ckyy, 2022-02-08:
@Fersy2003 yes with that one decoded, just use the csv file and it works now.

Fersy2003, 2022-02-08:
> @Fersy2003 yes with that one decoded, just use the csv file and it works now.

unfortunately it does not work for m
[image]
e

b14ckyy, 2022-02-08:
@Fersy2003 you have to use the latest development build that @avsaase has linked above.

Fersy2003, 2022-02-08:
> @Fersy2003 you have to use the latest development build that @avsaase has linked above.

ja, läuft nicht..

[LOG00004.01.zip](https://github.com/gyroflow/gyroflow/files/8023300/LOG00004.01.zip)
.

b14ckyy, 2022-02-08:
> > @Fersy2003 you have to use the latest development build that @avsaase has linked above.
> 
> ja, läuft nicht..
> 
> [LOG00004.01.zip](https://github.com/gyroflow/gyroflow/files/8023300/LOG00004.01.zip) .

What Blackbox Tools version do you use? With the latest one I see the same error. With the 0.4.7-alpha0 version it accepts the CSV but no traces now with the latest GF build. Need to check with the devs.

Fersy2003, 2022-02-08:
> What Blackbox Tools version do you use? With the latest one I see the same error. With the 0.4.7-alpha0 version it accepts the CSV but no traces now with the latest GF build. Need to check with the [...]

ahh ok, that explains it

ilya-epifanov, 2022-02-08:
The issue is fixed, no new reports are needed. Please wait for an update.

AdrianEddy, 2022-02-08:
Fixed in e532d34025823dacbeaf2e8fb90708b3df56a90c
Download here: https://github.com/gyroflow/gyroflow/suites/5220852577/artifacts/160018759

GarryBu, 2022-02-22:
Windows11
1-4RC
Inav 4.0.0

https://user-images.githubusercontent.com/60334441/155184772-0a32d0d8-d351-45d1-adae-c76f25420275.mp4

[LOG00007.zip](https://github.com/gyroflow/gyroflow/files/8118717/LOG00007.zip)