gyroflow/gyroflow #71: [Feature request]: Lens Profile "Runcam_5_Orange_4K_30FPS_XV_16by9_stretched" lens profile is not working correctly

Seba112PL, 2022-02-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Description

Is there an existing issue for this?
 I have searched the existing issues
Description
I've tried using lens profile attached with new Gyroflow 1.0 with XV video from Runcam but it appears that it no longer works correctly. Video is zoomed no matter what setting I change. Using normal profile "Runcam_5_orange_145_deg_2160p_16by9" but it distorts video a bit.

ElvinC, 2022-02-04:
Yep, that's one of the special presets with an additional horizontal stretching factor that v1.0.0-rc1 doesn't take into account yet, the feature will be added.

AdrianEddy, 2022-04-03:
Fixed in 151cc291a9daf3cbe68762759862a974aad2030c

closed 2022-04-03 by AdrianEddy