gyroflow/gyroflow #73: [Bug]: no stabilisation for Runcam 5 Orange

krl91, 2022-02-05:
### Gyroflow version

1.0 rc

### What operating system are you using?

last macos cersion

### What GPU are you using?

Intel

### What happened?

I'm using Runcam 5 Orange, I followed the documentation:
load video
load csv file
load lense profile
add synch (about 10) point
choose Lock horizon
fov about 2.8
select a window to render
then start the render.

no stabilization at all on rendered file

[image]

AdrianEddy, 2022-02-05:
Does the preview look good? stabilized? Does the same happen with other export formats?

ElvinC, 2022-02-05:
I have seen another user previously with what looks like the same problem. If you open the csv file in a text editor, is everything `0`? If so, it's an issue with the camera as opposed to Gyroflow.

krl91, 2022-02-05:
the preview is not good...  :(

krl91, 2022-02-05:
my bad all are set to zero in csv file

closed 2022-02-05 by krl91