gyroflow/gyroflow #74: [Bug]: very slow on MacBook m1 pro

bendi89, 2022-02-05:
### Gyroflow version

1.0 rc1 

### What operating system are you using?

OS X 

### What GPU are you using?

M1 pro 

### What happened?

Very slow on this platform. 2,5 hours to stabilize 3 min of 5k gopro 10 file.

AdrianEddy, 2022-02-05:
Which export codec? ProRes is not hardware accelerated, use x264 or x265

bendi89, 2022-02-05:
> Which export codec? ProRes is not hardware accelerated, use x264 or x265

I use x265

SebastianKumor, 2022-02-05:
@AdrianEddy I did experience similar result on intel macbook pro with radeon pro 5500m and x265. 4k 4:3 4 min video took 3 hours to stabilize. I have run the same clip trough reelsteady to test it and it was done under 1 hour.

AdrianEddy, 2022-02-12:
Fixed in rc3

closed 2022-02-12 by AdrianEddy

rcragger, 2022-02-22:
Seems to still be an issue on my Macbook Pro 16" with rc3. I suspect that Gyroflow is not switching graphics cards to use 5500m card.

JRNYC7, 2024-04-30:
I encountered the same issue where it took 3 hours to complete a task that should have taken 2 minutes, all because I was using internal memory. However, the speed increased to just 20 minutes when I switched to external storage.