gyroflow/gyroflow #76: [Bug]: Semantic versioning issues in Cargo.toml

atouchet, 2022-02-05:
### Gyroflow version

n/a

### What operating system are you using?

Windows 10

### What happened?

Crates should not be set as requiring version "*" in Cargo.toml as any breaking chances in a dependency can cause compatibility problems.

AdrianEddy, 2022-02-05:
I know, this will be sorted out before final v1.0.0 release :)

ilya-epifanov, 2022-02-08:
@atouchet @AdrianEddy it's not a big deal since it's not a library crate. as soon as there's a `Cargo.lock` and some testing, it's gonna be just fine. So no need to rush it before 1.0.0. It should definitely be done at some point, however

AdrianEddy, 2022-02-08:
Fixed in 4ccdb1685e6fff44c2d74c08a2bb110794ccdcd9

closed 2022-02-08 by AdrianEddy