gyroflow/gyroflow #77: [Bug]: missing OpenCV library

genarks, 2022-02-05:
### Gyroflow version

from source

### What operating system are you using?

Windows10

### What happened?

Thanks for the great software.

I built it on Windows.
The build procedure described worked, but in my environment, I had to add "opencv_objdetect" to the OpenCV library to be linked ($Env:OPENCV_LINK_LIBS in __env.ps1).

(In my environment, the OpenCV library was built without a version number such as "454".)

It may vary depending on the version of OpenCV, but I will let you know just in case.

B-nutze-RR, 2022-02-05:
I installed cuda 11.5 toolkit or use:
vcpkg install opencl:x64-windows-release

AdrianEddy, 2022-02-05:
probably need to remove version numbers from OPENCV_LINK_LIBRARIES in __env.ps1, I'll check that

AdrianEddy, 2022-02-06:
Fixed in c79e7c6d91cae2165fc3437da69bf77f9829eac0

closed 2022-02-06 by AdrianEddy