gyroflow/gyroflow #82: [Feature request]: Windows 7 support

mrslycat, 2022-02-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Description

Hi, thank you for this awesome project!
Unfortunately, the Qt 6 framework is no more compatible with Win7.
I've tried to build it using Qt 5.15.2 but get an error when compiling a library "qml-video-rs", which also relays on QT 6+.

Is there a reason why a Qt 6 was chosen and any chance the project can be ported on Qt 5 ?

AdrianEddy, 2022-02-06:
Yes, there is a reason Qt 6 was choosen: This is a new software and fairly advanced, so it's not designed to run on old or slow hardware. If I would choose to use Qt 5.15, next year it will be unsupported and I would have to port everything to Qt 6 anyway.

Is there are reason you still use Win 7 in 2022?

closed 2022-02-06 by AdrianEddy

Avispa, 2022-10-04:
@dncoder: what is your status? Did you get it compile on/for _Windows 7_?

What is the latest version which supports _Windows 7_, if any?

AdrianEddy, 2022-10-04:
There isn't one and it's not possible to get it to work on Windows 7 at this point, It's not supported by Qt (UI framework we use) and going back to previous Qt 5 is too much work at this point.

What's the reason to not upgrade to Windows 10?

Avispa, 2022-10-04:
Thanks for your, extremely quick, answer. However, I have to say, regarding:

> There isn't one and it's not possible to get it to work on Windows 7 at this point (...)

... 2 seconds of internet search proofs your claim wrong:
https://forum.qt.io/topic/133002/qt-creator-6-0-1-and-qt-6-2-2-running-on-windows-7

AdrianEddy, 2022-10-04:
If mean sure you can modify Qt to work on Windows 7 by removing the calls to new windows APIs but that requires you to recompile Qt yourself and it's not something I'm interesting in doing and maintaining.

I see someone posted dll files for pathed Qt 6.3.0 there. You can download original Qt 6.3.0 yourself, then these patched dlls, replace these provided dlls and then replace all Qt dlls in Gyroflow directory and that should work.

Gyroflow should work with 6.3.0 Qt if you just replace the dlls