gyroflow/gyroflow #83: [Bug]: ffmpeg error / nvcuda.dll missins

wesgtr, 2022-02-05:
### Gyroflow version

1.0.0

### What operating system are you using?

Windows 11

### What GPU are you using?

AMD RX 570

### What happened?

Whe Click on export receiver errors about ffmpeg error and missing nvcuda.dll

[image]

bigjoncoop, 2022-02-06:
Im no expert but if your running a AMD RX 570, that card isn't compatible with CUDA. only NVIDEA cards have CUDA cores...
But im pretty sure gyroflow also can work with OPENCL which AMD does support.

Curious to see what the DEVS have to say

wesgtr, 2022-02-06:
Yes. I used gyroflow 0.3.0 with this exactly specs

HerbertEttore, 2022-02-06:
I have the same hardware and I have this same problem.
I think the AMD AMF/VCE renderer specific to AMD GPU needs to be implemented.

CPU: Ryzen 5 3600
GPU: RX570 4gb
RAM: 16gb
OS: Windows 11

Gro2mi, 2022-02-07:
Same problem here with Intel Iris Xe GPU

junqixinyi, 2022-02-07:
My rx470 and 4800u integrated video card have the same problem.

AdrianEddy, 2022-02-07:
Please test this one and report back: https://github.com/gyroflow/gyroflow/suites/5205488855/artifacts/159064470

Gro2mi, 2022-02-07:
Working for Intel Iris XE

wesgtr, 2022-02-07:
@AdrianEddy Worked on AMD Radeon RX570

AdrianEddy, 2022-02-07:
ok so this one is fixed, closing it then

schlaakj, 2022-02-07:
I have the Same Problem here Ryzen 9 3900x amd 5700xt

AdrianEddy, 2022-02-07:
> I have the Same Problem here Ryzen 9 3900x amd 5700xt

tried this one? https://github.com/gyroflow/gyroflow/suites/5205488855/artifacts/159064470