gyroflow/gyroflow #86: [Bug]: instant crash when typing in low pass filter without Gyrofile selected

Williwolle, 2022-02-05:
### Gyroflow version

1.0.0

### What operating system are you using?

windows 11 pro 21H2

### What GPU are you using?

Nvidia GTX 1050

### What happened?

When I try to set a low pass filter under the section " Motion data", before having a Gyro log selected, the application instantly closes. It happens when I type in the value for the lowpass filter, not when I tick the lowpass filter box.

### Relevant log output

```shell
19:03:12 [WARN] Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt.
You must not let any exception whatsoever propagate through Qt code.
19:03:12 [ERROR] Object 0x197a12f72a0 destroyed while one of its QML signal handlers is in progress.
Most likely the object was deleted synchronously (use QObject::deleteLater() instead), or the application is running a nested event loop.
This behavior is NOT supported!
qrc:/src/ui/menu/MotionData.qml:126: function() { [native code] }
```

seanmavley, 2022-02-06:
Yep, came here to report same thing. Glad it's been reported already.

AdrianEddy, 2022-02-06:
Fixed in 8173d46cbb57fb768ac60cd43b7f620a25350521

closed 2022-02-06 by AdrianEddy