gyroflow/gyroflow #87: [Bug]: "freeze frames" in render

swoopyfpv, 2022-02-05:
### Gyroflow version

v1.0.0.rc1

### What operating system are you using?

Mac OSX Big Sur / 11.6

### What GPU are you using?

M1

### What happened?

render has a some "freeze frames"
preview seems to be fine / tried with gpu on/off same problem - occurs on some flips in the footage

camera & gyro from sony rx0ii (in this case with zeiss exo 0.6x lens)
footage auto synced / camera profile also auto generated

see problem here: https://youtu.be/G2RoQkhgJww

swoopyfpv, 2022-02-06:
edit:
also noticed no sound in consistent distance. (see red circle)
and fps is not like the original file (og files are 25fps) output files vary (see picture - at red line
[image]
)

AdrianEddy, 2022-02-15:
Does this work good with https://gyroflow.xyz/devbuild/?autodownload ? Audio issue is a separate one tracked in #146

swoopyfpv, 2022-02-15:
> Does this work good with https://gyroflow.xyz/devbuild/?autodownload ? Audio issue is a separate one tracked in #146

thanks for the quick reply.

unfortunately i still have glitches in the rendered file.
(could only launch when opening with terminal "contents/MacOs/gyroflow")

additional info:
- auto sync worked fine with rc1 & rc2 / with rc3 & the one you send it get's stuck syncing so i need to do manual points
- rendering since rc3 is faster and needs less performance

swoopyfpv, 2022-02-19:
> How about this one? https://gyroflow.xyz/devbuild/?autodownload
> 
> Also let me know if autosync works. Does it work when right-clicked on timeline -> `Auto sync here`? Does lens calibrator work?

video: https://www.youtube.com/watch?v=OzbsMlWAiuQ

- freeze frames still occur (video 0:00 - 0:28 )
- autosync shown from 0:29 - 0:44 (syncing with right-click in timeline works fine)
- just did a quick test: lens calibration on auto gives 0 good points / on windows the same file gives a few good frames (if you need video just comment)
- permission for folders is asked multiple times ( video 0:44 - end)

mycosd, 2022-02-19:
To narrow down this issue, can you test this on another System?
Can you provide us with the files so we can try reproducing it?

AdrianEddy, 2022-02-19:
Yes, please send us the file (both this flight and calibration one). Does this also happen with different export codecs? x264, x265 and ProRes?

swoopyfpv, 2022-02-19:
> Yes, please send us the file (both this flight and calibration one). Does this also happen with different export codecs? x264, x265 and ProRes?

happens with all codecs.
here is a video file: https://drive.google.com/file/d/1_3oiEP5d5vvgRytsKJ4oBia2kvBT41dp/view?usp=sharing
here is the lens file: https://drive.google.com/file/d/1D6tS1A3DXdpDbN5vT0sS3Gitw4B-4M9h/view?usp=sharing

mycosd, 2022-02-19:
I can't reproduce any of the export issues. With RC2 I can hear the audio dropping out issue, which is fixed already. It might be system related. Have you tried it with `No smoothing` & `No zooming` setting?
Since you also mentioned permission for folder issue... maybe the frame drop happen when those permission dialogs show up? No idea why they show up during rendering in the first place though.

swoopyfpv, 2022-02-19:
> I can't reproduce any of the export issues. With RC2 I can hear the audio dropping out issue, which is fixed already. It might be system related. Have you tried it with `No smoothing` & `No zooming` [...]

tried "no smoothing" & "no zooming" - worked fine.
now i also can't reproduce the freeze frames anymore... wtf :D
rendered two files who had the issue before, both fine now.

appreciate your help @AdrianEddy & @mycosd sry for the inconvenience

mycosd, 2022-02-19:
I'm happy for you that it's working... but it's not really a solution to the problem, as it might pop up again :)
It might be related to a specific smoothing/zooming algorithm... that's what we found out, I guess

swoopyfpv, 2022-02-19:
> I'm happy for you that it's working... but it's not really a solution to the problem, as it might pop up again :) It might be related to a specific smoothing/zooming algorithm... that's what we found out, I guess

if you got some specific things i could try feel free to tell me

AdrianEddy, 2022-04-30:
@swoopyfpv is everything working for you since then? Can this issue be closed?

swoopyfpv, 2022-04-30:
> @swoopyfpv is everything working for you since then? Can this issue be closed?

issue solved! thanks!

swoopyfpv, 2022-04-30:
> @swoopyfpv is everything working for you since then? Can this issue be closed?

issue solved. thanks!

closed 2022-04-30 by swoopyfpv