gyroflow/gyroflow #88: [Bug]: GyroFlow crash with a specific video from Runcam 5 Orange

krl91, 2022-02-05:
### Gyroflow version

1.0 RC

### What operating system are you using?

Windows 10

### What GPU are you using?

NVidia

### What happened?

Crash when I upload csv file after the video and custom lens from Runcam 5 Orange

All the file are here:
[files](https://drive.google.com/drive/folders/1NGCfUm6A9uYjRsNbnuoFMiZlsKLeakzQ?usp=sharing)

### Relevant log output

```shell
19:28:37 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\XXXX\\Desktop\\Gyroflow-windows64\\camera_presets"
19:28:37 [INFO] Loaded 1289 lens profiles in 71.833ms
19:28:37 [INFO] Latest version: v1.0.0-rc1, current version: v1.0.0-rc1
19:28:48 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
19:28:48 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "thread.video" ""
19:28:48 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
19:28:48 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
19:28:48 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
19:28:48 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
19:28:48 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x28906347c00) QSize(1640, 922)
19:28:48 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "decoder.video" "MFT"
19:28:48 [DEBUG] (13) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
19:28:48 [INFO] surface size: 1280x719
19:28:48 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce GTX 1080
19:28:48 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
19:28:48 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x28905c991c0) QSize(1280, 719)
19:28:48 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce GTX 1080
thread '<unnamed>' panicked at 'range end index 200 out of range for slice of length 0', C:\Users\runneradmin\.cargo\git\checkouts\telemetry-parser-19821e7ac6d09f51\802bb36\src\phone_apps\gf_recorder.rs:9:16
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Rayon: detected unexpected panic; aborting
```

krl91, 2022-02-05:
FYI it works well on macos ...

AdrianEddy, 2022-02-07:
I fixed it in telemetry-parser

closed 2022-02-07 by AdrianEddy

krl91, 2022-02-07:
> Unable to recreate with the latest code (Windows 10, Nvidia). Could you try again with this build when it is complete? https://github.com/gyroflow/gyroflow/actions/runs/1807000345

Hi, well done! thanks 
that correct the issue