gyroflow/gyroflow #89: [Bug]: Longer videos show no gyro traces until I zoom in

b14ckyy, 2022-02-05:
### Gyroflow version

1.0.0-rc1

### What operating system are you using?

Windows 11 - 22000.438

### What GPU are you using?

Core i7 7700K + GTX 1080ti

### What happened?

[image]
[image]

As seen in the pictures, it seems like if more than 20min of video is shown in the timeline, the Gyro traces disappear. Not sure if I have merged the original files back then or if the SMO 4K has no file size limit but this can be confusing. 

I guess sample file is not needed? If you need one I will upload the 21GB video :D

AdrianEddy, 2022-02-06:
Could you use `gyro2bb` and upload just the csv with the gyro data? https://github.com/AdrianEddy/telemetry-parser/releases

b14ckyy, 2022-02-06:
@AdrianEddy here you go: 
[PRO_VID_20210614_170412_00_002.mp4.zip](https://github.com/gyroflow/gyroflow/files/8010301/PRO_VID_20210614_170412_00_002.mp4.zip)

AdrianEddy, 2022-02-06:
Fixed in 8319d87742194dcfe17341d914ebfe6ec186706e

closed 2022-02-06 by AdrianEddy