gyroflow/gyroflow #92: [Bug]: CSV file unsupported

Jrham08, 2022-02-05:
### Gyroflow version

1.0.0rc

### What operating system are you using?

Windows 10

### What GPU are you using?

AMD Vega 56

### What happened?

Using Firefly x lite 4k , Gyroflow allows me to select my media and my lens profile but as soon as I try to apply my motion date with a csv file I receive an error unsupported file format.
[144055AA.CSV](https://github.com/gyroflow/gyroflow/files/8008967/144055AA.CSV)
[image]

ElvinC, 2022-02-06:
This is fixed with https://github.com/AdrianEddy/telemetry-parser/pull/9
Note that the firefly gyro data has been shown to contain gaps in the gyro data, so your results may vary.

closed 2022-02-06 by ElvinC

MarkfpvMark, 2022-02-06:
Hi! thanks! how can i install this update in gyoflow?

ElvinC, 2022-02-06:
Automated builds are in the `Actions` page. Direct link to latest one: https://github.com/gyroflow/gyroflow/suites/5196621997/artifacts/158496425