gyroflow/gyroflow #96: [Bug]: Either IMU orientation or rotation doesn't work propperly

KwadLife, 2022-02-05:
### Gyroflow version

1.0.0

### What operating system are you using?

Windows 10

### What GPU are you using?

Nvidia geforce GTX 1060

### What happened?

I am using a Naked Hero 6 flipped upside down on my Cinewhoop. 
Whatever I try to do I cannot get the footage stabelized. In RSGO I simply press flip gyro under the advanced settings and it works without a problem.  I have tried all kind of IMU orientation combinations XYZ, xYZ, XyZ, XYz etc...  and Roation Roll at 180° ... 
Auto Sync didnt work either...  Auto IMU and Rotation was set to ZyX P0 R0 Y0  

Small test footage can be found here https://1drv.ms/v/s!AtD8FclY_0WIgbcbmCJoDB0BLDux9Q?e=nDfEsm

[image]

ElvinC, 2022-02-06:
Looking at the file, seems like setting the Yaw value to 180 degrees gives the correct orientation and stabilization. This indicates there might be an bug in the rotation matrix generation causing roll and yaw to be swapped, specifically this part:

https://github.com/gyroflow/gyroflow/blob/master/src/core/gyro_source.rs#L227

AdrianEddy, 2022-02-06:
yes, it's possible the pitch/roll/yaw angles are mixed up, we need to figure out the correct order and fix this there in gyro_source.rs

AdrianEddy, 2022-02-07:
Fixed in 7e2818d08f0fac92f81f916fa3e983df19e739fd

closed 2022-02-07 by AdrianEddy

KwadLife, 2022-02-09:
fixed