gyroflow/gyroflow #97: [Bug]: Betaflight Blackbox not extracting gyro information

JonnIEnglish, 2022-02-05:
### Gyroflow version

 v1.0.0-rc1

### What operating system are you using?

MacOS Big Sur version 11.4

### What happened?

It seems that the gyro data is not being extracted from both Betaflight raw blackbox logs and converted CSV logs.
[image]

### Relevant log output

```shell
Last login: Sat Feb  5 17:46:38 on ttys000
jonathan@MacBook-Pro ~ % /Applications/Gyroflow.app/Contents/MacOS/gyroflow 
MDK 0.14.0 (git adff8b8) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: macOS1090 x86_64; AppleClang13.0.0 (clang-1300.0.29.30)GCC4.2.1; libc++12000; 16:45:48 Feb  1 2022
22:04:27 [WARN] Populating font family aliases took 905 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost. 
22:04:28 [INFO] Lens profiles directory: "/Applications/Gyroflow.app/Contents/Resources/camera_presets"
22:04:28 [INFO] Loaded 1289 lens profiles in 95.443ms
22:04:29 [INFO] Latest version: v1.0.0-rc1, current version: v1.0.0-rc1
22:05:00 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
22:05:00 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "thread.video" ""
22:05:00 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
22:05:00 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
22:05:00 [DEBUG] (12) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
22:05:00 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Metal
22:05:00 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7fa00be12270) QSize(1544, 822)
22:05:00 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "decoder.video" "VideoToolbox"
22:05:01 [DEBUG] (13) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
22:05:01 [INFO] surface size: 1352x720
AudioQueuePause @4349.000 (sample time)/36654148654333 (host time)
22:05:01 [INFO] OpenCL Platform: Apple, Device: Intel Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
AudioQueuePause @7248.000 (sample time)/36654230422308 (host time)
AudioQueuePause @14553.000 (sample time)/36654404024883 (host time)
22:05:01 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Metal
22:05:01 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7fa00dda2e90) QSize(1352, 720)
22:05:02 [INFO] OpenCL Platform: Apple, Device: Intel Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
AudioQueuePause @17474.000 (sample time)/36655124797038 (host time)
AudioQueuePause @24759.000 (sample time)/36655297987279 (host time)
22:05:31 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_VIDEOTOOLBOX 0
22:05:31 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_VIDEOTOOLBOX (videotoolbox) with format Some(AV_PIX_FMT_VIDEOTOOLBOX)

22:05:52 [INFO] No movement detected, max gyro angle: 1.3841417502467876. Skipping sync point.
22:05:52 [WARN] Sync point out of acceptable range 9998.990000000005 < 9000
22:05:52 [WARN] Sync point out of acceptable range 9998.990000000005 < 9000
22:05:52 [INFO] No movement detected, max gyro angle: 1.3841417502467876. Skipping sync point.
22:05:52 [WARN] Sync point out of acceptable range 9998.989999999998 < 9000
22:05:52 [WARN] Sync point out of acceptable range 9998.989999999998 < 9000
Invalid float 
Invalid float 
Invalid float 
Invalid float 
Invalid float 
22:06:20 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_VIDEOTOOLBOX 0
22:06:20 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_VIDEOTOOLBOX (videotoolbox) with format Some(AV_PIX_FMT_VIDEOTOOLBOX)

22:06:32 [INFO] No movement detected, max gyro angle: 1.3841417502467876. Skipping sync point.
22:06:32 [WARN] Sync point out of acceptable range 9998.990000000005 < 9000
22:06:33 [WARN] Sync point out of acceptable range 9998.990000000005 < 9000
22:06:33 [INFO] No movement detected, max gyro angle: 1.3841417502467876. Skipping sync point.
22:06:33 [WARN] Sync point out of acceptable range 9998.989999999998 < 9000
22:06:33 [WARN] Sync point out of acceptable range 9998.989999999998 < 9000
AudioQueuePause @32093.000 (sample time)/36765614460166 (host time)
AudioQueuePause @39419.000 (sample time)/36765951040368 (host time)
AudioQueuePause @46750.000 (sample time)/36766566371402 (host time)
AudioQueuePause @54081.000 (sample time)/36766775354001 (host time)
AudioQueuePause @61414.000 (sample time)/36766949529477 (host time)
AudioQueuePause @64320.000 (sample time)/36767170764908 (host time)
AudioQueuePause @67803.000 (sample time)/36767264707696 (host time)
AudioQueuePause @70662.000 (sample time)/36767345640449 (host time)
AudioQueuePause @75927.000 (sample time)/36767476738324 (host time)
AudioQueuePause @83028.000 (sample time)/36767646093133 (host time)
AudioQueuePause @89608.000 (sample time)/36768048387163 (host time)
AudioQueuePause @92357.000 (sample time)/36768127047535 (host time)
AudioQueuePause @99458.000 (sample time)/36768308006499 (host time)
AudioQueuePause @106783.000 (sample time)/36768505254119 (host time)
AudioQueuePause @114125.000 (sample time)/36768842115462 (host time)
AudioQueuePause @121445.000 (sample time)/36769027647421 (host time)
AudioQueuePause @127380.000 (sample time)/36769590648550 (host time)
AudioQueuePause @134436.000 (sample time)/36769793903252 (host time)
AudioQueuePause @141775.000 (sample time)/36770026299130 (host time)
AudioQueuePause @149099.000 (sample time)/36770513784968 (host time)
AudioQueuePause @156430.000 (sample time)/36770943335900 (host time)
```

AdrianEddy, 2022-02-05:
Please upload that file

JonnIEnglish, 2022-02-06:
Oh yes, of course. Here we go.

[Uploading LOG00002.BFL…]()

AdrianEddy, 2022-02-06:
Please try uploading again, looks like it didn't attach the file properly

BanniUK, 2022-02-06:
Blackbox log thats crashing gyroflow https://drive.google.com/file/d/1hXexVFmY7fsgiAxTZXCcfzYPNjhx5oUp/view?usp=sharing

JonnIEnglish, 2022-02-07:
Apologies, here we are.

[LOG00002.BFL.zip](https://github.com/gyroflow/gyroflow/files/8013162/LOG00002.BFL.zip)

leitmanr, 2022-02-07:
I have one that forces the GF to crash as well
[btfl_001.zip](https://github.com/gyroflow/gyroflow/files/8014723/btfl_001.zip)

nerdCopter, 2022-02-07:
two BBL that do not show gyro data as well: 
https://drive.google.com/drive/folders/1c3leS2WLJUSX_eysfpZWHs1FVcRr7LF5?usp=sharing