gyroflow/gyroflow #98: [Bug]: FireFly xLite csv file not supported

MarkfpvMark, 2022-02-05:
### Gyroflow version

v1

### What operating system are you using?

Os Mojave 10.14.6 

### What GPU are you using?

Intel Iris

### What happened?

[image]

hello,
when i try to load the csv file from the firefly xLite camera this message showup, is impossible load the gyro data.

ElvinC, 2022-02-06:
Could you attach an example file?

ElvinC, 2022-02-06:
Duplicate of https://github.com/gyroflow/gyroflow/issues/98

closed 2022-02-06 by ElvinC

MarkfpvMark, 2022-02-06:
Sure here the csv file link download
[
](https://we.tl/t-3kjAILb5jH)