gyroflow/gyroflow #103: [Bug]: Changing Output resolution changes Safe area

Seba112PL, 2022-02-06:
### Gyroflow version

1.0

### What operating system are you using?

Windows 10

### What happened?

When using gyroflow 1.0 and when changing output resolution of file it changes safe area causing it to zoom too much or too little (showing egdes) despite using dynamic croping

It is happening on Insta 360 go2 and on Runcam 5 orange (propably on more cameras)

[image]
[image]
[image]

redstar01, 2022-02-06:
Hi! Tested on BetaFPV SMO 4K camera and got the same behavior

Seba112PL, 2022-02-07:
Hero 6. Same thing

[image]
[image]
[image]

Seba112PL, 2022-02-07:
Hero 7. same thing

[image]
[image]
[image]
g

redstar01, 2022-02-08:
#113 #110

AdrianEddy, 2022-02-08:
> Not sure but it seems not related to lens profiles, same cases #113 #110. I guess the problem is more general

these are two different issues, we'll fix both

AdrianEddy, 2022-02-08:
it also happens with this lens profile
[Panasonic_BGH1_Laowa7.5mm_3840x2160_16by9.json.zip](https://github.com/gyroflow/gyroflow/files/8027127/Panasonic_BGH1_Laowa7.5mm_3840x2160_16by9.json.zip)

marcroe, 2022-02-09:
I think I know why this is happening. The border of the image frame is approximated/interpolated with a few points that are determined using undistort operation. The "grid" for interpolation is not dense enough for those very curved profiles (insta360). But I am a bit surprised why this happens with GoPro footage. I have not experienced this. There might be another issue that's linked to the FOV slider

AdrianEddy, 2022-02-09:
That Hero 7 lens profile is definitely wrong, I would just ignore that one. Also the Hero6 comment was about changing the output resolution and that's fixed too. 

So the remaining issue is with Insta360 lens profiles, and that BGH1 one I posted.
I'm also sometimes seeing tiny black borders on sharp moves in certain videos, something is not entirely accurate there

marcroe, 2022-02-10:
Ok, great. then it's definitely the interpolation of the frame border that's causing it.
Will look into it.

mycosd, 2022-02-20:
I've tested it with #201 and it doesn't appear to let the safe area go outside the video, like the default FOV estimator. However it isn't correct either. The problem is, the boundary polygon of the go2 lens isn't continous, it has spots in it, which both estimation processes don't expect at all
[image]

AdrianEddy, 2022-05-04:
This is no longer an issue for correct lens profiles

closed 2022-05-04 by AdrianEddy