gyroflow/gyroflow #104: [Bug]: Missing Polar Nano preset

jesuslg123, 2022-02-06:
### Gyroflow version

1.0.0-rc1

### What operating system are you using?

Mac OS 12.0.1

### What happened?

Hello,

On the latest version 1.0.0-rc1 Polar Nano preset is missing, which it was available already. I can see there are 2 different polar preset as well.

[image]

AdrianEddy, 2022-02-06:
You mean it was present in 0.3.0? I don't see it here: https://github.com/ElvinC/gyroflow/tree/master/camera_presets/Caddx

AdrianEddy, 2022-02-06:
Fixed in 1ac0da28ea582229ea324b39b8d2a661545d4303

closed 2022-02-06 by AdrianEddy

jesuslg123, 2022-02-06:
Great! Thanks I just realized it was in another fork and also on the dev branch https://github.com/ElvinC/gyroflow/pull/131 
To many changes but for great! Awesome work! Version 1 looks stunning!