gyroflow/gyroflow #107: [Bug]: i can not export video (ffmpeg error) , with or without gpu encoding

Fersy2003, 2022-02-06:
### Gyroflow version

1.0.0-rc1

### What operating system are you using?

Win11

### What GPU are you using?

RTX2080TI

### What happened?

when exporting, the following error message is displayed:

22:48:44 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_DXVA2 0
22:48:44 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_DXVA2 (dxva2) with format Some(AV_PIX_FMT_DXVA2_VLD)

22:48:44 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: Some("dxva2")
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
22:48:44 [DEBUG] (2) gyroflow::rendering: proc.video_codec: Some("hevc_nvenc")
22:48:44 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 294133, render_frame_count: 17648
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Get transfer formats from GPU, frame.is_null: false, hw_ctx.is_null: false, format: DXVA2_VLD
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: [AV_PIX_FMT_NV12], frame.is_empty: true
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_CUDA), encoder_pixel_format: None
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: DXVA2_VLD, color range: MPEG
22:48:44 [WARN] Encoder doesn't support the desired pixel format (AV_PIX_FMT_DXVA2_VLD)

22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: dev.sw_formats: [AV_PIX_FMT_NV12, AV_PIX_FMT_YUV420P, AV_PIX_FMT_YUVA420P, AV_PIX_FMT_YUV444P, AV_PIX_FMT_P010LE, AV_PIX_FMT_P016LE, AV_PIX_FMT_YUV444P16LE, AV_PIX_FMT_BGR0, AV_PIX_FMT_RGB0]
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: target_format: AV_PIX_FMT_NV12
[NULL @ 000002185C981080] No codec provided to avcodec_open2()
22:48:44 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
22:48:44 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_NONE () with format None

22:48:44 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: None
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
22:48:44 [DEBUG] (2) gyroflow::rendering: proc.video_codec: Some("hevc_nvenc")
22:48:44 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 294133, render_frame_count: 17648
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_CUDA), encoder_pixel_format: None
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: YUV420P, color range: MPEG
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: target_format: AV_PIX_FMT_YUV420P
[NULL @ 000002185C771AC0] No codec provided to avcodec_open2()
22:48:44 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_D3D11VA 2
22:48:44 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_D3D11VA (d3d11va) with format Some(AV_PIX_FMT_D3D11)

22:48:44 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: Some("d3d11va")
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
22:48:44 [DEBUG] (2) gyroflow::rendering: proc.video_codec: Some("hevc_nvenc")
22:48:44 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 294133, render_frame_count: 17648
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Get transfer formats from GPU, frame.is_null: false, hw_ctx.is_null: false, format: D3D11
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: [AV_PIX_FMT_NV12], frame.is_empty: false
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_CUDA), encoder_pixel_format: None
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: D3D11, color range: MPEG
22:48:44 [WARN] Encoder doesn't support the desired pixel format (AV_PIX_FMT_D3D11)

22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: dev.sw_formats: [AV_PIX_FMT_NV12, AV_PIX_FMT_YUV420P, AV_PIX_FMT_YUVA420P, AV_PIX_FMT_YUV444P, AV_PIX_FMT_P010LE, AV_PIX_FMT_P016LE, AV_PIX_FMT_YUV444P16LE, AV_PIX_FMT_BGR0, AV_PIX_FMT_RGB0]
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: target_format: AV_PIX_FMT_NV12
[NULL @ 000002185C770D40] No codec provided to avcodec_open2()
22:48:44 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_CUDA 3
22:48:44 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_CUDA (cuda) with format Some(AV_PIX_FMT_CUDA)

22:48:44 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: Some("cuda")
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
22:48:44 [DEBUG] (2) gyroflow::rendering: proc.video_codec: Some("hevc_nvenc")
[... 22 lines cut ...]
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 15
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 16
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 17
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 18
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 19
22:48:44 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_NONE () with format None

22:48:44 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: None
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
22:48:44 [DEBUG] (2) gyroflow::rendering: proc.video_codec: Some("hevc_nvenc")
22:48:44 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 294133, render_frame_count: 17648
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_CUDA), encoder_pixel_format: None
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: YUV420P, color range: MPEG
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: target_format: AV_PIX_FMT_YUV420P
[NULL @ 000002185C772D00] No codec provided to avcodec_open2()
22:48:44 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
22:48:44 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: None
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
22:48:44 [DEBUG] (2) gyroflow::rendering: proc.video_codec: Some("hevc_nvenc")
22:48:44 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 294133, render_frame_count: 17648
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_CUDA), encoder_pixel_format: None
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: YUV420P, color range: MPEG
22:48:44 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: target_format: AV_PIX_FMT_YUV420P
[NULL @ 000002185C7711C0] No codec provided to avcodec_open2()
[image]

AdrianEddy, 2022-02-07:
Please try with this version: https://github.com/gyroflow/gyroflow/suites/5205488855/artifacts/159064470

Fersy2003, 2022-02-07:
the error still exists

19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: dev.sw_formats: [AV_PIX_FMT_NV12, AV_PIX_FMT_YUV420P, AV_PIX_FMT_YUVA420P, AV_PIX_FMT_YUV444P, AV_PIX_FMT_P010LE, AV_PIX_FMT_P016LE, AV_PIX_FMT_YUV444P16LE, AV_PIX_FMT_BGR0, AV_PIX_FMT_RGB0]
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: target_format: AV_PIX_FMT_NV12
[NULL @ 0000020987D7D9C0] No codec provided to avcodec_open2()
19:57:00 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 4
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 5
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 6
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 7
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 8
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 9
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 10
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 11
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 12
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 13
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 14
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 15
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 16
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 17
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 18
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 19
19:57:00 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_NONE () with format None

19:57:00 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: None
19:57:00 [DEBUG] (2) gyroflow::rendering: Possible encoders with NVIDIA: [("hevc_nvenc", true), ("nvenc_hevc", true), ("hevc_mf", true), ("hevc_qsv", true), ("libx265", false)]
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
19:57:00 [DEBUG] (2) gyroflow::rendering: proc.video_codec: Some("hevc_nvenc")
19:57:00 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 303967, render_frame_count: 18238
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_CUDA), encoder_pixel_format: None
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: YUV420P, color range: MPEG
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: target_format: AV_PIX_FMT_YUV420P
[NULL @ 0000020987D7D500] No codec provided to avcodec_open2()
19:57:00 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
19:57:00 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: None
19:57:00 [DEBUG] (2) gyroflow::rendering: Possible encoders with NVIDIA: [("hevc_nvenc", true), ("nvenc_hevc", true), ("hevc_mf", true), ("hevc_qsv", true), ("libx265", false)]
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
19:57:00 [DEBUG] (2) gyroflow::rendering: proc.video_codec: Some("hevc_nvenc")
19:57:00 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 303967, render_frame_count: 18238
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_CUDA), encoder_pixel_format: None
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: YUV420P, color range: MPEG
19:57:00 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: target_format: AV_PIX_FMT_YUV420P
[NULL @ 0000020987D7CC00] No codec provided to avcodec_open2()

[BTFL_BLACKBOX_LOG_Titan_DC5_20211212_180846.BBL_video0000.zip](https://github.com/gyroflow/gyroflow/files/8018316/BTFL_BLACKBOX_LOG_Titan_DC5_20211212_180846.BBL_video0000.zip)

Fersy2003, 2022-02-07:
I have found that the problem is not with the black box data but with the video file.

it looks like it only affects videos from the airunit, others go without problems...

I have uploaded an example video with the error occurs:
https://drive.google.com/file/d/17Cc4XDxg0nZ4U044TPEydqQxg0XrsagR/view?usp=sharing

kwkwkwkw, 2022-02-08:
Same Problem here error when exporting video and seems no stabilization ever made even successfully applied the auto sync the gyro data

kwkwkwkw, 2022-02-08:
> Please try with this version: https://github.com/gyroflow/gyroflow/suites/5205488855/artifacts/159064470

Tried and the program shutdown automatically while trying to start to vendering

AdrianEddy, 2022-02-08:
what is your video file? recording from dji air unit?

Fersy2003, 2022-02-08:
> what is your video file? recording from dji air unit?

the error affects all my video taken with the airunit on sd card. the same video from the dji goggles works against it .

just load and export the video file, lens profile and motion data are not needed for this error.

try my video:
https://drive.google.com/file/d/1lBlQc6zzlJIJFEDtqTF9Quh6b52qNDJU/view?usp=sharing

[image]

kwkwkwkw, 2022-02-08:
> what is your video file? recording from dji air unit?

yes, from air unit and the gyro data from the onboard SD card but I figured out the stabilization issue was from the exported gyroflow file created by the previous version 1.0.0 RC1.  With the one you suggested to try and using the original csv / bfl data file, it works well for the stabilization preview but still will shutdown the program when hitting the video export button.

AdrianEddy, 2022-02-08:
Known issue with videos from air unit, for now you can try transcoding first

kwkwkwkw, 2022-02-08:
> yes, from air unit and the gyro data from the onboard SD card but I figured out the stabilization issue was from the exported gyroflow file created by the previous version 1.0.0 RC1. With the one [...]

Oh !  Just find that it can't load the gyro data completely and correctly most of the time and cause the issue of stabilization.

Fersy2003, 2022-02-08:
> Known issue with videos from air unit, for now you can try transcoding first

i am not in a hurry, just trying to help troubleshoot ;)

kwkwkwkw, 2022-02-08:
> Known issue with videos from air unit, for now you can try transcoding first

What should I do with that ?  Use any video editor to convert the video file first you mean ?

But I can make it with the same files set in the beta version !

kwkwkwkw, 2022-02-08:
> Oh ! Just find that it can't load the gyro data completely and correctly most of the time and cause the issue of stabilization.

Sorry !  May be I mixed up the 2 versions' path and starting the wrong version on the last test !

kwkwkwkw, 2022-02-08:
Great !  I can export the video now after re-vendered the original DJI output file with video editor even with a high bitrate format.

But only can output the video under non-GPU rendering mode, it will still be shutted down if using the GPU function.

My old MSI notebook is running dual display cards, one is the onboard Intel(R) HD Graphics 4600
and the other one is NVIDIA GeForce GTX 780M.

AdrianEddy, 2022-02-08:
please run Gyroflow_with_console.bat and paste the log output here. and make sure you have latest dev build https://gyroflow.xyz/devbuild/?autodownload

AdrianEddy, 2022-02-08:
Fixed in c444d3df0f3d01cb8a1e29cfe857613778a0e62a

closed 2022-02-08 by AdrianEddy

Fersy2003, 2022-02-09:
> Fixed in [c444d3d](https://github.com/gyroflow/gyroflow/commit/c444d3df0f3d01cb8a1e29cfe857613778a0e62a)

it works, great!!!