gyroflow/gyroflow #99: [Bug]: Amd export error

Ilianleboss, 2022-02-06:
### Gyroflow version

v1.0.0-rc1

### What operating system are you using?

Windows 10

### What GPU are you using?

Amd Rx550 4gb

### What happened?

Hello!
While exporting the video with my GPU (AMD Rx550 4gb), I have an error in the terminal and the video does not export, I post it here maybe you could fix the error in the next update or help me😉

### Relevant log output

```shell
09:37:19 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\ilian\\Downloads\\Gyroflow-windows64\\camera_presets"
09:37:19 [INFO] Loaded 1289 lens profiles in 18.196ms
09:37:19 [INFO] Latest version: v1.0.0-rc1, current version: v1.0.0-rc1
09:37:21 [DEBUG] (8) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
09:37:21 [DEBUG] (9) MDKPlayer: [in m_player->onEvent] "thread.video" ""
09:37:21 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
09:37:21 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
09:37:21 [DEBUG] (10) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
09:37:21 [DEBUG] (11) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
09:37:21 [DEBUG] (11) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x1fffef19050) QSize(1030, 579)
09:37:21 [DEBUG] (9) MDKPlayer: [in m_player->onEvent] "decoder.video" "MFT"
09:37:21 [DEBUG] (11) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
09:37:21 [INFO] surface size: 959x719
09:37:21 [DEBUG] (11) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
09:37:21 [DEBUG] (11) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x1fff9a0edd0) QSize(959, 719)
09:37:22 [INFO] OpenCL Platform: AMD Accelerated Parallel Processing, Device: Advanced Micro Devices, Inc. gfx803
09:37:22 [INFO] OpenCL Platform: AMD Accelerated Parallel Processing, Device: Advanced Micro Devices, Inc. gfx803
09:37:22 [INFO] OpenCL Platform: AMD Accelerated Parallel Processing, Device: Advanced Micro Devices, Inc. gfx803
09:37:22 [INFO] OpenCL Platform: AMD Accelerated Parallel Processing, Device: Advanced Micro Devices, Inc. gfx803
09:37:22 [DEBUG] (1) qml: Setting param smoothness_pitch 0.2
09:37:22 [DEBUG] (1) qml: Setting param smoothness_yaw 0.2
09:37:22 [DEBUG] (1) qml: Setting param smoothness_roll 0.2
09:37:28 [DEBUG] (2) gyroflow::rendering: ffmpeg_hw::supported_gpu_backends: ["cuda", "dxva2", "qsv", "d3d11va"]
09:37:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [dec] codec type AV_HWDEVICE_TYPE_DXVA2 0
09:37:28 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_DXVA2 (dxva2) with format Some(AV_PIX_FMT_DXVA2_VLD)

09:37:28 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: Some("dxva2")
09:37:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
09:37:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
[AVHWDeviceContext @ 000001FF896C44C0] Cannot load nvcuda.dll
[AVHWDeviceContext @ 000001FF896C44C0] Could not dynamically load CUDA
09:37:28 [DEBUG] (2) gyroflow::rendering: Failed to create specified HW device: AV_HWDEVICE_TYPE_CUDA

09:37:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 1, for: hevc_nvenc
09:37:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
[AVHWDeviceContext @ 000001FFFF7599C0] Cannot load nvcuda.dll
[AVHWDeviceContext @ 000001FFFF7599C0] Could not dynamically load CUDA
09:37:28 [DEBUG] (2) gyroflow::rendering: Failed to create specified HW device: AV_HWDEVICE_TYPE_CUDA

09:37:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_D3D11VA 2, for: hevc_nvenc
09:37:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_D3D11VA
09:37:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_D3D11VA
09:37:28 [DEBUG] (2) gyroflow::rendering: proc.video_codec: Some("hevc_nvenc")
09:37:28 [DEBUG] (2) gyroflow::rendering: start_us: 0, render_duration: 175075, render_frame_count: 5247
09:37:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Get transfer formats from GPU, frame.is_null: false, hw_ctx.is_null: false, format: DXVA2_VLD
09:37:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Hardware transfer formats from GPU: [AV_PIX_FMT_NV12], frame.is_empty: true
09:37:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Codec formats: [AV_PIX_FMT_YUV420P, AV_PIX_FMT_NV12, AV_PIX_FMT_P010LE, AV_PIX_FMT_YUV444P, AV_PIX_FMT_P016LE, AV_PIX_FMT_YUV444P16LE, AV_PIX_FMT_BGR0, AV_PIX_FMT_RGB0, AV_PIX_FMT_CUDA, AV_PIX_FMT_D3D11]
09:37:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: hw_device_type: Some(AV_HWDEVICE_TYPE_D3D11VA), encoder_pixel_format: Some(NV12)
09:37:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_video: Setting output pixel format: NV12, color range: JPEG
09:37:28 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: target_format: AV_PIX_FMT_NV12
[AVHWFramesContext @ 000001FF8D6D30C0] Could not create the texture (80070057)
09:37:28 [DEBUG] (2) gyroflow::rendering: Failed to initialize frame context. Error code: -1313558101

09:37:28 [DEBUG] (2) gyroflow::rendering: Failed to create encoder HW context.

[hevc_nvenc @ 000001FF8D977000] Cannot load nvcuda.dll
[hevc_nvenc @ 000001FF8D977000] cuda_load_functions error
```

wesgtr, 2022-02-06:
How do you get the error on terminal? I tested and now opened terminal however I'm having error with Rx 570

Ilianleboss, 2022-02-07:
I open Gyroflow with Gyroflow_with_console.bat in the Gyroflow folder.

AdrianEddy, 2022-02-07:
Fixed in 6276bc4c98b0a2db5e8cea78de5bdfe937aaed61

closed 2022-02-07 by AdrianEddy