gyroflow/gyroflow #113: [Bug]: FOV changing output size

NehatLi, 2022-02-07:
### Gyroflow version

1.0.0-rc1

### What operating system are you using?

Windows 10

### What happened?

When I change the output size, I get a tiny view like if I set the fov greater than 1. To get the "right" view and export, I need to set the fov to something like 0.3-0.33 and it's really annoying.
I don't know if it's normal or not.

AdrianEddy, 2022-02-08:
Fixed in 0eb75f87a41e70594c234ea889ec73b758a8cd7f

closed 2022-02-08 by AdrianEddy

avsaase, 2022-02-08:
Awesome. One minor point: the current zoom percentage in the lower-left corner of the preview is not adjusted correctly. The max zoom value is correct.

NehatLi, 2022-02-08:
> Fixed in [0eb75f8](https://github.com/gyroflow/gyroflow/commit/0eb75f87a41e70594c234ea889ec73b758a8cd7f)

How do i update my gyroflow to get the update?

AdrianEddy, 2022-02-08:
https://gyroflow.xyz/devbuild/?autodownload or go to Github actions and download the last successful one

AdrianEddy, 2022-02-08:
> Awesome. One minor point: the current zoom percentage in the lower-left corner of the preview is not adjusted correctly. The max zoom value is correct.

This is intentional, you can then see actual zoom in relation to your input, so if you have 30% you can see that it'll be downscaled by this much. Do you think it makes sense to provide both (scaled and unscaled) zoom values?

avsaase, 2022-02-08:
> This is intentional, you can then see actual zoom in relation to your input, so if you have 30% you can see that it'll be downscaled by this much. Do you think it makes sense to provide both (scaled and unscaled) zoom values?

The way I understood it the zoom percentage represents the zoom required to keep the black parts out of the video.  So when it's 100% there is no additional zoom. I think this is also what the max zoom value represents.

avsaase, 2022-02-08:
That interpretation only holds when FOV is set to 1 but I do think it is the most straightforward way to think about it in the context of dynamic zoom.