gyroflow/gyroflow #114: [Bug]: Use GPU encoding: error: Ca not load nvcuda.dll

faalbers, 2022-02-07:
### Gyroflow version

v1.0.0-rc1

### What operating system are you using?

Windows 10

### What GPU are you using?

ADM Radeon Pro 460

### What happened?

ffmpeg:
"Cannot load nvcuda.dll"
"Could not dynamically load CUDA"

### Relevant log output

```shell
17:39:56 [DEBUG] (2) gyroflow::rendering: proc.gpu_device: Some("dxva2")
17:39:56 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: [enc] codec type AV_HWDEVICE_TYPE_CUDA 0, for: hevc_nvenc
17:39:56 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
[AVHWDeviceContext @ 0000020BD30AA700] Cannot load nvcuda.dll
[AVHWDeviceContext @ 0000020BD30AA700] Could not dynamically load CUDA
17:39:56 [DEBUG] (2) gyroflow::rendering: Failed to create specified HW device: AV_HWDEVICE_TYPE_CUDA
```

AdrianEddy, 2022-02-07:
Try this one and let me know https://github.com/gyroflow/gyroflow/suites/5205488855/artifacts/159064470

faalbers, 2022-02-07:
No errors now. There was no speed improvement with GPU though. It was even a few seconds slower.

AdrianEddy, 2022-02-08:
https://github.com/gyroflow/gyroflow/suites/5212720294/artifacts/159493241 this one should be better on AMD

faalbers, 2022-02-08:
It is WAY faster now , but the file's Bitrate is only 11 Mbps instead of the 57 Mbps as set in the settings gui

AdrianEddy, 2022-02-08:
please post output from [MediaInfo](https://mediaarea.net/en/MediaInfo)

AdrianEddy, 2022-02-08:
Is the filesize the same or also much smaller?

faalbers, 2022-02-09:
No GPU video info:
https://drive.google.com/file/d/1zoj1BSmUzaCGBphT_JW2WQoLVmdrzpk_/view?usp=sharing

GPU video info:
https://drive.google.com/file/d/18_dEyn0qOllXugEjoY_PMg886SCBuiep/view?usp=sharing

faalbers, 2022-02-09:
Also, as expected, the no GPU output has pixelformat = yuvj420p. Same as the GoPro footage.
But the GPU file has pixelformat = yuv420p , which is a smaller color range. This can introduce banding in areas with very small color change like blue sky horizons and such. RSGO has the same problem

AdrianEddy, 2022-02-19:
@faalbers Please check this issue with latest dev build https://gyroflow.xyz/devbuild/?autodownload

faalbers, 2022-02-21:
Still the same with GPU. 19 mb/s and yuv420p

AdrianEddy, 2022-02-21:
Can you send me original file as well as both encoded with GPU and without GPU?
Did you compare colors? format is reported differently but it should be the same, because of "color range" which is set separately

AdrianEddy, 2022-05-03:
Could you check the bitrate issue in the latest dev build? https://gyroflow.xyz/devbuild/?autodownload