gyroflow/gyroflow #117: [Bug]:

Williwolle, 2022-02-07:
### Gyroflow version

1.0.0(gh158)

### What operating system are you using?

windows 11 pro 21H2

### What GPU are you using?

Nvidia GTX 1050

### What happened?

Its a bit difficult to explain but ok:
 I have a csv file imported to Gyroflow and I got everything setup: (low pass filter and orientation for Camera angle)
If I then decide to load in an new Video + Gyrofile or just a Gyrofile (can be the same as before to) the program seems to automaticly have applied the same low pass + orientation as before, but actually it didn´t apply them, eveen though a low pass filter and the orientation seems to be applied if you look at the interface. If have to reenter the low pass filter + orientation to take effect in the Gyrochart.
This was very confusing to me and gave me a lot of headache.
I hope you can fix this soon, but great Program overall, thank you!

(I know its a bit difficult to understand if there are any questions feel free to ask them.)

### Relevant log output

```shell
There wasn't any error or warning displayed
```

Williwolle, 2022-02-07:
Sorry didn't want to take away your title, we edited it at the same time...

ElvinC, 2022-02-08:
Fixed in f1aa0767db0ed91b39f5d6c4d7019e6671882b5b
Now it applies the previous settings at load (useful for a bunch of similar files)

closed 2022-02-08 by ElvinC