gyroflow/gyroflow #120: [Bug]: UI issues : In multi-byte languages, the display lines will be misaligned.

inouekou, 2022-02-08:
### Gyroflow version

v1.0.0-RC1 (gh169)

### What operating system are you using?

Windows 10

### What happened?

In multi-byte languages, the display lines will be misaligned.

[image]

inouekou, 2022-02-08:
It is only in the left pane that the lines are misaligned.

[image]

AdrianEddy, 2022-02-08:
Should be fixed in this one: https://github.com/gyroflow/gyroflow/suites/5222869367/artifacts/160142556

closed 2022-02-08 by AdrianEddy