gyroflow/gyroflow #123: [Feature request]:  Caddx Orca

Logicenios, 2022-02-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Description

Hello everyone,

I have been using the program since alpha and i can say the latest release is mind-bending for the progress you have made. The request is that if we can support caddx orca (which i have) i know it has eis which means there is some type of gyro inside. I have also contacted Orca if they can release any firmware with gyro output.

Thanks for your time and amazing work,
S.G.

ElvinC, 2022-02-08:
I was in contact with Caddx last month and know they are looking at Gyro logging already (not sure if it's for the Orca). There's nothing we can do on our side at the moment.

Logicenios, 2022-02-08:
They have some type of holiday so i should receive an update on the upcoming days.
I will update the issue if i have an answer.

Logicenios, 2022-02-11:
Hello again, 

I have got an answer from them that the camera does not have a gyro...

Thanks,
S.G.

closed 2022-02-11 by ElvinC