gyroflow/gyroflow #124: [Bug]: App is often unresponsive on a Mac Mini 2018

Sjullep, 2022-02-08:
### Gyroflow version

v1.0.0-rc1 (gh186)

### What operating system are you using?

macOS Monterey 12.1 

### What GPU are you using?

Intel UHD Graphics 630

### What happened?

Sometimes the program reacts immideately after clicking or keyboard input, sometimes it takes up to 5-10 minutes (or even longer)  to respond. Nothing happens in the terminal screen, no errors. 
CPU load is mostly low, but sometimes (after opening the video file) very high for 5-15 minutes or longer
All the sliders are working however, and i can preview the video too. I have 64GB of RAM.
Exporting works without any errors, GPU encoding checked

### Relevant log output

```shell
leon@MacMini ~ % /Applications/_DRONE/Gyroflow\ 1.0.0\ gh186.app/Contents/MacOS/gyroflow ; exit;
MDK 0.14.0 (git 1f8761a) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: macOS1090 x86_64; AppleClang13.0.0 (clang-1300.0.29.30)GCC4.2.1; libc++12000; 12:19:11 Feb  6 2022
12:37:37 [WARN] Populating font family aliases took 180 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost. 
12:37:38 [INFO] Lens profiles directory: "/Applications/_DRONE/Gyroflow 1.0.0 gh186.app/Contents/Resources/camera_presets"
12:37:38 [INFO] Loaded 1290 lens profiles in 11.740ms
12:37:38 [INFO] OpenCL Platform: Apple, Device: Intel Intel(R) Core(TM) i7-8700B CPU @ 3.20GHz
12:37:38 [WARN] Unknown GPU intel intel(r) core(tm) i7-8700b cpu @ 3.20ghz
12:37:38 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_VIDEOTOOLBOX
12:37:38 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_VIDEOTOOLBOX
[src/rendering/mod.rs:45] gpu_type = Unknown
12:37:38 [INFO] Latest version: v1.0.0-rc1, current version: v1.0.0-rc1
```

AdrianEddy, 2022-02-10:
Please check this with latest build: https://gyroflow.xyz/devbuild/?autodownload

Sjullep, 2022-02-10:
Sorry, i notice no difference. For example when i start the program an load i videofile, the CPU goes max, and when i click the button to load a lens profile, nothing happens. When i want to quit the program, and klick on "Yes" in the popup, nothing happens.
But in this version the preview window seems to work, i can now play the video. In former versions this wasn't possible. 
So all the buttons and checkboxes are non responsive, but all the sliders are ok and working: when i move the slider FOV for example, the image changes. When i click on the "Play" button above the timeline, nothing happens, but when i hit the spacebar, the video starts to play.
And mostly i have a high CPU load when this happens

[image]

AdrianEddy, 2022-04-28:
Does this still happen to you in latest dev build? https://gyroflow.xyz/devbuild/?autodownload

Sjullep, 2022-04-29:
Yes, it still happens. Has somethinh to do with my Wacom tablet. When i use the mouse, everything is normal. But when i use the Wacom tablet, i have to use the right-click buttom to get any response

Sjullep, 2022-05-02:
> This is definitely related to Wacom tablet, looks like it's an issue in Qt,
> Can you test this:
> Right click on Gyroflow -> Open Package Contents -> Contents -> MacOS
> Open Terminal and drag&drop gyroflow file to the terminal
> Add -nowmpointer after gyroflow in terminal
> Hit enter
> so the command line becomes something like this: /Applications/Gyroflow/Contents/MacOS/gyroflow -nonowmpointer

I tried this, but it soen's make a difference, buttons are still unresponsive.
Here's the output of the terminal:

***@***.*** ~ % /Applications/_DRONE/Gyroflow/Gyroflow\ RC5\ GH459.app/Contents/MacOS/gyroflow -nowmpointer 
MDK 0.14.1 (git 2addef3) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: macOS1090 x86_64; AppleClang13.0.0 (clang-1300.0.29.30)GCC4.2.1; libc++12000; 13:01:16 Apr 28 2022
14:31:39 [WARN] Populating font family aliases took 174 ms. Replace uses of missing font family "Roboto" with one that exists to avoid this cost. 
14:31:39 [WARN] Binding on animation is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
14:31:39 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
14:31:39 [WARN] Binding on popup is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
14:31:39 [WARN] Binding on indicator is not deferred as requested by the DeferredPropertyNames class info because one or more of its sub-objects contain an id.
14:31:39 [WARN] Binding on contentItem is not deferred as requested by the DeferredPropertyNames class info because it constitutes a group property.
14:31:40 [INFO] Lens profiles directory: "/Applications/_DRONE/Gyroflow/Gyroflow RC5 GH459.app/Contents/Resources/camera_presets"
14:31:40 [INFO] Loaded 1605 lens profiles in 14.441ms
14:31:40 [DEBUG] (1) gyroflow_core::gpu::opencl: OpenCL devices: Ok("Apple") ["Intel(R) Core(TM) i7-8700B CPU @ 3.20GHz", "Intel(R) UHD Graphics 630"]
14:31:40 [INFO] OpenCL Platform: Apple, Device: Intel Inc. Intel(R) UHD Graphics 630
[src/rendering/mod.rs:49] gpu_type = Intel
14:31:40 [INFO] Latest version: v1.0.0-rc5, current version: 1.0.0-rc5 (gh459)

AdrianEddy, 2022-09-30:
Could you please test latest dev build? Qt was updated to new version so I'm curious if that fixed it https://gyroflow.xyz/devbuild/?autodownload

Sjullep, 2022-10-02:
Hi Adrian, I haven't had time to test it very thoroughly but everything seems to work OK now. 
I am curious how you solved it, since i am experiencing the same problems with Ultimaker-Cura version 5.1, and perhaps the solution you found will also work for Cura

AdrianEddy, 2022-10-02:
Thanks for testing, I'm closing this issue then.
I didn't do anything on my side, I updated Qt version to 6.4.0, I can see it regularly gets bugs and patches related to Wacom tablet, so one of these probably fixed it.

Looks like it was this one? https://bugreports.qt.io/browse/QTBUG-98936?jql=text%20~%20%22wacom%22

closed 2022-10-02 by AdrianEddy

Sjullep, 2022-10-11:
I noticed that Ultimaker Cura 5.0.0 Beta shows the same problems with the wacom tablet, like unresponsive buttons