gyroflow/gyroflow #134: Linux 1.0rc1 appimage missing features

eburrows, 2022-02-08:
The Linux 1.0rc1 appimage is missing at least:
1. Video rotation adjustment under the "Video Information" pane. The pencil icon does not appear.
2. Export aspect ratio unlocking. The padlock icon does not appear.

Observed on:
Ubuntu 18.04
nVidia GTX 1080, with nvidia drivers (not noveau)
OpenCL using CUDA
Insta360 GO2 with updated firmware Pro-Wide source file

AdrianEddy, 2022-02-08:
Do all other icons appear? can you post screenshot?

eburrows, 2022-02-08:
Screenshot attached
[image]
.

AdrianEddy, 2022-02-08:
ok, all icons are missing, I saw this already on some linux distros. Probably some dependency is missing, they are svg icons

eburrows, 2022-02-09:
I installed the listed dependencies, and I don't see any obvious missing SVG-related packages from Apt. 

I can run the AppImage, but not the binary, as I have glibc6 2.27, not the required 2.28. Perhaps it is some dependency on Ubuntu 20. Unfortunately I can't run strace on the appimage, so I can't see what SVG libraries it's trying to load. I'll see if there's a way to get glibc6 2.28 available, or run it in a VM.

eburrows, 2022-02-09:
When I run the appimage as root with strace, I can see it trying to load /usr/share/pixmaps/pencil.svg, but that's not on my system. I assume it's not in the appimage either.

AdrianEddy, 2022-02-09:
It's also not our icons. Our icons are stored in binary as resource so they won't be loaded from the filesystem. The issue is probably with svg renderer so look for some shared library failing to load

eburrows, 2022-02-09:
The only SVG library I see failing to load is plugins/imageformats/libqsvg.so.avx2.

$ grep -i svg strace.out | grep -v 'usr/share'
23734 writev(5, [{iov_base="X\0\0\0\0\0\0\0\302\r\0\0\0\0\0\0", iov_len=16}, {iov_base="<\t\25,\0\0\0\0\"\0\0\0\0\0\0\0\16\0\0\0\10\0\0\0libqsvgi"..., iov_len=72}], 2) = 88
23731 lstat("/tmp/.mount_Gyrofl8EOQ6s/plugins/iconengines/libqsvgicon.so",  <unfinished ...>
23731 openat(AT_FDCWD, "/tmp/.mount_Gyrofl8EOQ6s/plugins/iconengines/libqsvgicon.so", O_RDONLY|O_CLOEXEC <unfinished ...>
23734 writev(5, [{iov_base="X\0\0\0\0\0\0\0\337\r\0\0\0\0\0\0", iov_len=16}, {iov_base="\234\t\25,\0\0\0\0\36\0\0\0\0\0\0\0\n\0\0\0\10\0\0\0libqsvg."..., iov_len=72}], 2) = 88
23731 lstat("/tmp/.mount_Gyrofl8EOQ6s/plugins/imageformats/libqsvg.so",  <unfinished ...>
23731 openat(AT_FDCWD, "/tmp/.mount_Gyrofl8EOQ6s/plugins/imageformats/libqsvg.so", O_RDONLY|O_CLOEXEC <unfinished ...>
23731 openat(AT_FDCWD, "/tmp/.mount_Gyrofl8EOQ6s/plugins/imageformats/libqsvg.so.avx2", O_RDONLY|O_CLOEXEC <unfinished ...>
23731 access("/tmp/.mount_Gyrofl8EOQ6s/plugins/imageformats/libqsvg.so.avx2", F_OK <unfinished ...>
23731 openat(AT_FDCWD, "/tmp/.mount_Gyrofl8EOQ6s/plugins/imageformats/libqsvg.so", O_RDONLY|O_CLOEXEC <unfinished ...>
23731 openat(AT_FDCWD, "/tmp/.mount_Gyrofl8EOQ6s//lib/x86_64/libQt6Svg.so.6", O_RDONLY|O_CLOEXEC <unfinished ...>
23731 openat(AT_FDCWD, "/tmp/.mount_Gyrofl8EOQ6s//lib/libQt6Svg.so.6", O_RDONLY|O_CLOEXEC <unfinished ...>
23731 read(50, "Qt6Svg.so.6\n7f1565047000-7f15650"..., 1024) = 1024

$ cd /tmp/.mount_GyroflF0ZID7/
$ ls plugins/iconengines/libqsvgicon.so
plugins/iconengines/libqsvgicon.so
$ ls plugins/imageformats/libqsvg.so
plugins/imageformats/libqsvg.so
$ ls plugins/imageformats/libqsvg.so
plugins/imageformats/libqsvg.so
$ ls plugins/imageformats/libqsvg.so.avx2
ls: cannot access 'plugins/imageformats/libqsvg.so.avx2': No such file or directory
$ ls lib/x86_64/libQt6Svg.so.6
ls: cannot access 'lib/x86_64/libQt6Svg.so.6': No such file or directory
$ ls lib/libQt6Svg.so.6
lib/libQt6Svg.so.6

schnupperm, 2022-02-13:
I have the same problem on kubuntu 20.04 LTS and 21.10 with the tar.gz version.
strace output is attached. 
[strace.gyroflow.gz](https://github.com/gyroflow/gyroflow/files/8056455/strace.gyroflow.gz)

Entropy512, 2022-02-18:
Also happens for kubuntu 21.10 built from source, latest git master

It is interesting that @eburrows saw a load attempt for /usr/share/pixmaps/pencil.svg - it almost looks like something is attempting to load the icons from /usr/share/pixmaps instead of wherever they are embedded

`sudo ln -s ~adodd/gitrepos/gyroflow/resources/icons/svg/*.svg .` run in /usr/share/pixmaps will make the icons appear...  obviously NOT a valid solution.

Well most of the icons.  The file dialog is missing all icons so it's nearly impossible to use - just a white bar at the top.

schnupperm, 2022-02-23:
build from todays source on kubutu 20.04. Still no icons, even with Entorpy512s soft link work around.

pitts-mo, 2022-03-09:
I have also noticed open file dialogs do not appear to render correctly for me with linux appimages.
Is that symptom related to this issue?
-p

image is from v1.0.0-rc5 on lubuntu 20.04 w/intel cpu/gpu.

[image]

Entropy512, 2022-03-09:
Not just the appimages - also affects self-built on Kubuntu 21.10

JoshuaNitschke, 2022-07-24:
I'm taking a look at this issue.

JoshuaNitschke, 2022-07-24:
I made some progress, will try to finish it by next weekend.
[image]

commit 92004783 (2022-07-26): Fix missing icons on Linux - Fixes #134 (#410) 

Add backup icon.source in case theme fails to load
https://doc.qt.io/qt-6/qtquickcontrols2-icons.html
"Both icon.name and icon.source can be set to ensure that an icon will always be found. If the icon is found in the theme, it will always be used; even if icon.source is also set."
closed 2022-07-26 by AdrianEddy