gyroflow/gyroflow #135: Error on export

ih8unot, 2022-02-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

Latest

### What operating system are you using?

Windows 11

### What GPU are you using?

Geoforce 2070 Super

### What happened?

Error when attempting to export. Resolved by simple export to D:\
[image]

AdrianEddy, 2022-02-12:
[image]

AdrianEddy, 2022-02-15:
Fixed in 4aa9097f508ae0184f1fa6e483689532f1ba9644

closed 2022-02-15 by AdrianEddy