gyroflow/gyroflow #137: Unable to get platform id list after 10 seconds of waiting

hexarch, 2022-02-09:
lino@desktopos:~/git/gyroflow$ cargo run --release
    Finished release [optimized] target(s) in 0.12s
     Running `target/release/gyroflow`
MDK 0.14.0 (git 378299c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang13.0.1 GCC4.2.1; libc++13000; 15:11:36 Feb  1 2022
00:37:33 [INFO] Lens profiles directory: "/home/lino/git/gyroflow/resources/camera_presets"
00:37:33 [INFO] Loaded 1290 lens profiles in 9.525ms
00:37:35 [INFO] Latest version: v1.0.0-rc2, current version: 1.0.0-rc2 (dev3083)
thread 'main' panicked at 'Platform::default(): Unable to get platform id list after 10 seconds of waiting.', /home/lino/.cargo/registry/src/github.com-1ecc6299db9ec823/ocl-0.19.3/src/standard/platform.rs:177:55
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
lino@desktopos:~/git/gyroflow$ 

I'm on PopOS with Linux desktopos 5.15.15-76051515-generic #202201160435~1642693824~21.10~97db1bb SMP Thu Jan 20 17:35:05 U x86_64 x86_64 x86_64 GNU/Linux

AdrianEddy, 2022-02-09:
We'll fix it so it doesn't about, but you can try installing latest graphic card drivers or opencl package for your distro

hexarch, 2022-02-09:
Well, cannot install amdgpu drivers in Pop!OS because it only installs on ubuntu|mint|debian.
opencl says 0 number of platforms.
Some strange tips https://www.reddit.com/r/pop_os/comments/rjdfbq/my_amdgpupro_success_story_on_pop_os_2110_opencl/ where you change "pop" to "ubuntu"... no. I will have to put this on another machine, it's a strange territory and I don't want my machine unbootable. I hope it still boots, haven't tested.
Thanks

adrianalin, 2022-02-09:
Had the same problem in docker, maybe try installing `libpocl2`,

JoeriCapens, 2022-02-09:
I had the same error message on Debian 11. Fixed with "sudo apt install libpocl2", thank you @adrianalin !

pitts-mo, 2022-02-09:
I appear to be hitting the same issue with v1.0.0-rc2 appimage in ubuntu. However, the rc1 appimage seems to run ok.  My attempt to add libpocl2 has reached this version mismatch on a dependency.

sudo apt-get install libpocl2 clang-9 libclang-common-9-dev lib32gcc-s1 lib32stdc++6 libc6-i386 libc6
~
The following packages have unmet dependencies:
 libc6-i386 : Depends: libc6 (= 2.31-0ubuntu9.2) but 2.31-0ubuntu9.3 is to be installed
E: Unable to correct problems, you have held broken packages.

AdrianEddy, 2022-02-09:
Fixed the crash in 78ce4534c7499a5b411cd65cd5a6dfd3bcb35a44
Download: https://gyroflow.xyz/devbuild/?autodownload

closed 2022-02-09 by AdrianEddy

titulebolide, 2022-06-20:
I've experienced the same issue on v1.1.0, on Fedora 35.
Was fixed by installing the package `pocl`, which is ubuntu's `libpocl2`

```
thread '<unnamed>' panicked at 'Platform::list: Error retrieving platform list: Unable to get platform id list after 10 seconds of waiting.', /root/.cargo/registry/src/github.com-1ecc6299db9ec823/ocl-0.19.3/src/standard/platform.rs:50:14
```