gyroflow/gyroflow #140: autosync error: ffmpeg::Error(1313558101:

Schadi001, 2022-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

v1.0.0-rc2

### What operating system are you using?

MacOS Monerey 12.0.1

### What GPU are you using?

M1

### What happened?

Autosync is also not working with the new Gyroflow version [v1.0.0-rc2]
[image]

### Relevant log output

```shell
default	11:03:33.627052+0300	gyroflow	           AQMEIO_HAL.cpp:1751  FetchAndLogFrameSize GetCurrentIOBufferFrameSize deviceID=45, frame size is 512
default	11:03:33.631438+0300	gyroflow	SpatializationManager.cpp:391   binding = Default-Output
default	11:03:33.633270+0300	gyroflow	SpatializationManager.cpp:464   Audio route is built-in speakers.
default	11:03:33.633512+0300	gyroflow	SpatializationManager.cpp:615   Spatial preferences: prefersHeadTrackedSpatialization=0, prefersLossyAudioSources=0, maxSpatializableChannels=16, spatialAudioSourceCount=1, alwaysSpatialize=0, contentType='moov', overrideSpatialMode=1, nowPlayingEligible=1, spatialAudioSources=[ 'mlti' ]
default	11:03:33.633528+0300	gyroflow	       MEMixerChannel.cpp:1587  1-channel audiovisual content is NOT eligible for spatialization
default	11:03:33.633677+0300	gyroflow	           AQMEIO_HAL.cpp:1336  error 2003332927 setting kBluetoothAudioDevicePropertySpatialAudioStreamInfo.
default	11:03:33.633696+0300	gyroflow	       MEMixerChannel.cpp:1613  mFormatID='lpcm', mNumChannels=1, mBestAvailableContentType=0, mContentspatializable=0, mSpatializationStatus=0, err=2003332927
error	11:03:33.633707+0300	gyroflow	       MEMixerChannel.cpp:1639  client <AudioQueueObject@0x11e031e00; [0]; play> got error 2003332927 while sending format information
default	11:03:33.633732+0300	gyroflow	 MEDeviceStreamClient.cpp:429   AQME Default-Output: client stopping: <AudioQueueObject@0x11e031e00; [0]; play>; running count now 0
```

Schadi001, 2022-02-09:
default	11:03:22.174604+0300	runningboardd	Acquiring assertion targeting [app<application.xyz.gyroflow.3621328.3624670(501)>:65164] from originator [daemon<com.apple.WindowServer(88)>:137] with description <RBSAssertionDescriptor| "AppDrawing" ID:146-137-1492954 target:65164 attributes:[
	<RBSDomainAttribute| domain:"com.apple.appnap" name:"AppDrawing" sourceEnvironment:"(null)">,
	<RBSAcquisitionCompletionAttribute| policy:AfterApplication>
	]>
default	11:03:22.174675+0300	runningboardd	Assertion 146-137-1492954 (target:[app<application.xyz.gyroflow.3621328.3624670(501)>:65164]) will be created as active
default	11:03:22.175719+0300	runningboardd	[app<application.xyz.gyroflow.3621328.3624670(501)>:65164] Ignoring jetsam update because this process is not memory-managed
default	11:03:22.175739+0300	runningboardd	[app<application.xyz.gyroflow.3621328.3624670(501)>:65164] Ignoring suspend because this process is not lifecycle managed
default	11:03:22.175751+0300	runningboardd	[app<application.xyz.gyroflow.3621328.3624670(501)>:65164] Ignoring GPU update because this process is not GPU managed
default	11:03:22.175800+0300	runningboardd	Calculated state for app<application.xyz.gyroflow.3621328.3624670(501)>: running-active (role: UserInteractiveNonFocal)
default	11:03:22.175793+0300	runningboardd	[app<application.xyz.gyroflow.3621328.3624670(501)>:65164] Set AppNap state: <RBMutableProcessAppNapState|0x139a43360 enabled:Y active:N socket:N disk:N priority:N cpu:N timer:Tier0>
default	11:03:22.177596+0300	runningboardd	Acquiring assertion targeting [app<application.xyz.gyroflow.3621328.3624670(501)>:65164] from originator [app<application.xyz.gyroflow.3621328.3624670(501)>:65164] with description <RBSAssertionDescriptor| "AppNap adapter assertion" ID:146-65164-1492955 target:65164 attributes:[
	<RBSAcquisitionCompletionAttribute| policy:AfterApplication>,
	<RBSDomainAttribute| domain:"com.apple.appnap" name:"Enable" sourceEnvironment:"(null)">,
	<RBSDomainAttribute| domain:"com.apple.appnap" name:"Inactive" sourceEnvironment:"(null)">,
	<RBSDomainAttribute| domain:"com.apple.appnap" name:"PreventDiskThrottle" sourceEnvironment:"(null)">,
	<RBSDomainAttribute| domain:"com.apple.appnap" name:"PreventSuppressedCPU" sourceEnvironment:"(null)">,
	<RBSDomainAttribute| domain:"com.apple.appnap" name:"PreventLowPriorirtyCPU" sourceEnvironment:"(null)">,
	<RBSDomainAttribute| domain:"com.apple.appnap" name:"PreventBackgroundSockets" sourceEnvironment:"(null)">,
	<RBSDomainAttribute| domain:"com.apple.appnap" name:"PreventTimerThrottleTier0" sourceEnvironment:"(null)">
	]>
default	11:03:22.177891+0300	runningboardd	Assertion 146-65164-1492955 (target:[app<application.xyz.gyroflow.3621328.3624670(501)>:65164]) will be created as active
default	11:03:22.178641+0300	runningboardd	[app<application.xyz.gyroflow.3621328.3624670(501)>:65164] Ignoring jetsam update because this process is not memory-managed
default	11:03:22.178683+0300	runningboardd	[app<application.xyz.gyroflow.3621328.3624670(501)>:65164] Ignoring suspend because this process is not lifecycle managed
default	11:03:22.178514+0300	runningboardd	Calculated state for app<application.xyz.gyroflow.3621328.3624670(501)>: running-active (role: UserInteractiveNonFocal)
default	11:03:22.178734+0300	runningboardd	[app<application.xyz.gyroflow.3621328.3624670(501)>:65164] Ignoring GPU update because this process is not GPU managed
default	11:03:22.179465+0300	runningboardd	Invalidating assertion 146-65164-1492799 (target:[app<application.xyz.gyroflow.3621328.3624670(501)>:65164]) from originator [app<application.xyz.gyroflow.3621328.3624670(501)>:65164]
default	11:03:22.180485+0300	runningboardd	Acquiring assertion targeting [xpcservice<com.apple.appkit.xpc.openAndSavePanelService([app<application.xyz.gyroflow.3621328.3624670(501)>:65164])(501)>:65279:65279] from originator [xpcservice<com.apple.appkit.xpc.openAndSavePanelService([app<application.xyz.gyroflow.3621328.3624670(501)>:65164])(501)>:65279:65279] with description <RBSAssertionDescriptor| "AppNap adapter assertion" ID:146-65279-1492956 target:65279 attributes:[
	<RBSAcquisitionCompletionAttribute| policy:AfterApplication>,
	<RBSDomainAttribute| domain:"com.apple.appnap" name:"Enable" sourceEnvironment:"(null)">,
	<RBSDomainAttribute| domain:"com.apple.appnap" name:"Inactive" sourceEnvironment:"(null)">,
	<RBSDomainAttribute| domain:"com.apple.appnap" name:"PreventDiskThrottle" sourceEnvironment:"(null)">,
	<RBSDomainAttribute| domain:"com.apple.appnap" name:"PreventSuppressedCPU" sourceEnvironment:"(null)">,
	<RBSDomainAttribute| domain:"com.apple.appnap" name:"PreventLowPriorirtyCPU" sourceEnvironment:"(null)">,
	<RBSDomainAttribute| domain:"com.apple.appnap" name:"PreventBackgrou<…>
default	11:03:22.180748+0300	runningboardd	Assertion 146-65279-1492956 (target:[xpcservice<com.apple.appkit.xpc.openAndSavePanelService([app<application.xyz.gyroflow.3621328.3624670(501)>:65164])(501)>:65279:65279]) will be created as active
default	11:03:22.181700+0300	runningboardd	[xpcservice<com.apple.appkit.xpc.openAndSavePanelService([app<application.xyz.gyroflow.3621328.3624670(501)>:65164])(501)>:65279:65279] Ignoring jetsam update because this process is not memory-managed
default	11:03:22.181757+0300	runningboardd	[xpcservice<com.apple.appkit.xpc.openAndSavePanelService([app<application.xyz.gyroflow.3621328.3624670(501)>:65164])(501)>:65279:65279] Ignoring suspend because this process is not lifecycle managed
[... cut ...]
default	11:03:27.245618+0300	CAReportingService	    CAReportingService.mm:466   sessions for client 156 : (
    "reporterID=670014898400,serviceName=audiodeviceusage,ownerPid=156,clientProcessName=(null),HostApplicationDisplayID=xyz.gyroflow,started=1,singleMessageSession=0"
)
default	11:03:27.245834+0300	CAReportingService	    CAReportingService.mm:601   1 active reporting session(s) (
    "reporterID=670014898400,serviceName=audiodeviceusage,ownerPid=156,clientProcessName=(null),HostApplicationDisplayID=xyz.gyroflow,started=1,singleMessageSession=0"
)

tanguyfpv, 2022-02-09:
i have the same bug

AdrianEddy, 2022-02-09:
What happens if you uncheck `Advanced` -> `Use GPU decoding`?

AdrianEddy, 2022-02-12:
This should be fixed in 1.0.0-rc3

closed 2022-02-12 by AdrianEddy