gyroflow/gyroflow #141: Audio Desync on Insta360 One R 1-Inch

KwadLife, 2022-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0 RC2

### What operating system are you using?

Windows 10

### What GPU are you using?

Nvidia geforce GTX 1060

### What happened?

Exported Insta360 OneR 1 Inch edition footage and the Audio is heavily off. 
Used trim function so I am not sure if this is the same issue as bug #19

B-nutze-RR, 2022-02-09:
There should be no issue if you don't use the IN trim or export the entire clip

AdrianEddy, 2022-02-09:
Yes, it's the same as #19 , the amount of offset depends on  the size of audio packets (*groups of samples*), so it will vary between files

closed 2022-02-09 by AdrianEddy