gyroflow/gyroflow #142: Motion rate smoothing

orent, 2022-02-09:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

Apply time warping to smooth the rate of panning or camera motion.

Obviously not useful for video with sound or moving objects, but it could be very useful for sweeping pans and cameras flying through spaces.

A jerky pan from a low quality tripod could be converted to perfect pan, with a short smooth transition from stationary to a fixed pan rate and then down to back to stationary with a transition that takes exactly the same time.

A trajectory through space can be converted to follow exactly the same path but at a fixed speed or perhaps follow a selectable relationship such as speed inversely related to turn rate.

ElvinC, 2022-02-09:
As I understand it, you're asking for these things, correct?

**3D translational motion tracking and correction**
Determine the spatial position of the camera in order to correct for panning motion. The issue and major challenge here is that IMU Data cannot be reliably double-integrated to the get the spatial coordinate due to numerical instabilities (The error grows quadratically over time). This would only work with a more precise motion track, e.g. using VFX motion tracking software. You can have a look at the technical details of Microsoft Hyperlapse to see what this involves: https://www.youtube.com/watch?v=sA4Za3Hv6ng

**Dynamic speed change**
Dynamically change the speed of a pan such that a panning motion looks smooth. I see a number of issues with this approach, mainly the visual quality of the result. The result of speeding up or slowing down footage is stuttery results with inconsistent clip lengths. Pretty sure the time jumps will be very noticeable if you're filming anything moving. Combined with the the challenge of spatial motion tracking, not sure if this is practical. One approach would be strictly removing frames that stray too far from the "center" of the stabilization, but then the challenge becomes deciding which frames to remove and which to keep.

Something similar to this would be implementing https://github.com/gyroflow/gyroflow/issues/37 and animating the parameter of "fixed camera" smoothing method, which makes it possible to create a constant speed pan.

Note that the example of the low quality tripod is a pure rotation (assuming insignificant offset to the camera). This should already work well with the current software.

Let me know what you think

ilya-epifanov, 2022-02-09:
Regarding dynamic speed change, you can already do it with the openfx plugin and an appropriate video editor. It's quite easy to do with davinci resolve for example.

orent, 2022-02-09:
IIUC, the software currently performs no dynamic adjustment of the time dimension, only a geometric adjustment of frames,  based on smoothed geometric parameters extracted from multiple frames (with possible extra data from IMU). 

I am trying to find what new features would be made possible by adjusting both the geometric and time dimensions. First, it would obviously require interpolating new frames. General time warping is difficult and requires motion tracking of individual objects. But this kind of effect would be mostly useful for footage where the camera is the only thing moving - the result would be awkward with moving objects, anyway. So generating a fractional frame would only require interpolating the motion, pan and zoom of the two closest neighboring frames and applying a global correction to one of them (the one producing a better result, usually but not always the nearest neighbor).

The motion analysis already produces an estimate of the camera motion vector. If you apply a time warp inversely proportional to the absolute value of this vector it should normalize the motion between interpolated output frames to segments of the same direction as the original vector but with normalizrd absolute magnitude- the same path but with constant speed along the path. I don’t see why this should have any issues with integration of position. 

This should be useful, for example for handheld footage where the cameraman slows down at some points. The result will appear as if they were gliding at a fixed speed through space. Pan is just a simple special case where the camera position is stationary.

I’m sure there would be “interesting” interactions between spatial and temporal smoothing that would need to be worked out. And when the camera slows down too much this effect would need to be reduced       because the inverse of no motion is infinite time warping.

ElvinC, 2022-02-09:
> So generating a fractional frame would only require interpolating the motion, pan and zoom of the two closest neighboring frames and applying a global correction to one of them (the one producing a better result, usually but not always the nearest neighbor).

Ah so an alternative way to look at this problem would be, instead of smoothing the frames vs time, we start by defining a orientation trajectory with a certain path and sample rate, and for each sample, we find the closest frame in the source video and apply the relevant transformation. This is possible, but I'm not sure the complexity of implementing this would be worth it, especially considering all the limitations (only stationary camera, no moving subjects). It would require redoing a lot of the code since it's the "opposite" of the current approach. I think you would be better off looking creating a panorama from the video frames, and panning through the result.

> I don’t see why this should have any issues with integration of position.

Not sure what you're previous knowledge of motion tracking is, so apologies if I go over something you know already. TL;DR: This is not practically possible from the low-cost sensor in the general case.
Compact MEMS inertial measurement units can contain 
* Gyroscope, detects angular rate (deg/s) in xyz
* Accelerometer, acceleration (m/s^2) in xyz
* Magnetometer, magnetic field vector (gauss) in xyz

For "slow" handheld motion, accel and magn can provide an decent (but noisy) estimation of the absolute pitch/yaw/roll, while the gyroscope measures the rotational velocity. The gyroscope doesn't have a "reference" orientation, so by itself, has no idea where it started (Equivalent to the constant of integration). With the right software, combining these can give a decent (within a few degrees) orientation.

For the 3D position, we can only measure the *change in the change of the position* meaning we have no reference for either the initial position or speed. Suppose we're only doing motion in 1D and are using a sensor that's accurate to within 0.2 m/s^2. After being physically stationary for 1 minute, your estimated state is:
`velocity = 0.2 * 60 = 12 m/s` and `position = 0.5 * 0.2 * 60^2 = 360 m`. In the 3D case, the sensor accceleration has to be corrected by the previously computed rotational orientation, which adds another degree of error. In general, double integration  with well-calibrated MEMS sensors is only feasible for very short timeframes. And you still need to spend an hour on precise calibration.

There are some ways of getting better tracking over a longer period of time, but these all rely on assumpsions. For instance "The motion is cyclical with zero velocity at direction change", or "the sensor is on a person taking steps" If you're interested this video illustrates some of these points and a special example: https://www.youtube.com/watch?v=IroLp5VOPDE but this method won't work for general motion.