gyroflow/gyroflow #146: Jerky sound after export

WizzX-FPV, 2022-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0-rc2

### What operating system are you using?

Win 10

### What GPU are you using?

intel UHD 620 / GTX 1080

### What happened?

after exporting video, sound is noticeable jerky like from train. seems like audio is stretch as well (not synced to video

https://user-images.githubusercontent.com/40316777/153210113-0da9a55b-3627-4df2-aaef-fe7e1496b4d9.mp4

[image]
)

wvansl, 2022-02-09:
I have something similar but for me, the audio is still quite OK but every 1 or 2s there is a glitch sound.

AdrianEddy, 2022-02-09:
Please upload the source video, I'll check
@WizzX-FPV  and @wvansl

WizzX-FPV, 2022-02-10:
here you can find raw video (camera Runcam Split 3 nano) and also there's gyro log with rough offset 7.9s
https://mega.nz/folder/XkkUEaIK#uC8rDJx3WlAgHDOy5W8aNA

WizzX-FPV, 2022-02-15:
> Could you please check in this version? https://github.com/gyroflow/gyroflow/suites/5298111892/artifacts/164378495

I tried with this version - same problem with the sound

AdrianEddy, 2022-02-19:
Fixed by @mycosd in 168afb9200632ad35772bf9722482fd4679da000
Latest dev build: https://gyroflow.xyz/devbuild/?autodownload

closed 2022-02-19 by AdrianEddy