gyroflow/gyroflow #147: Wit motion logger graph disappears after auto sync

panzeromnom, 2022-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

RC2

### What operating system are you using?

Win 10 professional

### What GPU are you using?

AMD Radeon R9 380

### What happened?

I opened the Video file i was going to stabilize, then opened the WIT motion file (WIT5.txt).
Everything looked fine.

The graph clearly showed, that the recorded Gyro data was shorter than the Video file.

[image]

I hit Autosync with the following parameters selected: 2,0s offset, 10,00s sample size, 10 sample points.

After the Autosync was completed, it showed motiontracing points in the video feed and the resulting graphs.

But...

the Gyrodata from the Gyrosensor was split up into parts and it looked like some data was missing.

[image]

Resulting in not stabilized footage.

Looking foraward to the fix ;)

Greetings from Germany!

### Relevant log output

```shell
13:29:21 [INFO] Latest version: v1.0.0-rc2, current version: 1.0.0-rc2
13:29:21 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\David\\Desktop\\Gyroflow\\camera_presets"
13:29:21 [INFO] Loaded 1290 lens profiles in 27.718ms
13:29:22 [INFO] OpenCL Platform: AMD Accelerated Parallel Processing, Device: Advanced Micro Devices, Inc. Tonga
13:29:22 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_D3D11VA
13:29:22 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_D3D11VA
[src\rendering\mod.rs:45] gpu_type = AMD
13:29:29 [DEBUG] (14) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
13:29:29 [DEBUG] (15) MDKPlayer: [in m_player->onEvent] "thread.video" ""
13:29:29 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
13:29:29 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
13:29:29 [DEBUG] (16) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
13:29:29 [DEBUG] (17) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
13:29:29 [DEBUG] (17) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x18b43986970) QSize(1030, 543)
13:29:29 [DEBUG] (15) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.D3D11"
13:29:30 [DEBUG] (17) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
13:29:30 [INFO] surface size: 1365x719
13:29:30 [DEBUG] (17) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
13:29:30 [DEBUG] (17) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x18b45b524e0) QSize(1365, 719)
13:29:41 [DEBUG] (2) gyroflow_core::smoothing::velocity_dampened: Max velocity: 0.20256801849922523
13:29:41 [DEBUG] (2) gyroflow_core::smoothing::velocity_dampened: Max velocity: 0.20256801849922523
13:29:41 [DEBUG] (2) gyroflow_core::smoothing::velocity_dampened: Max velocity: 0.20256801849922523
[... 400 lines cut ...]
13:30:49 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
13:30:49 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
13:30:49 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
13:30:49 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
13:30:49 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
13:30:49 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
13:30:49 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
13:30:49 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
13:30:49 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
13:30:49 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
13:30:49 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
13:30:49 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
13:30:49 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
13:30:49 [ERROR] OpenCV error: Error { code: 0, message: "Model not found" }
13:30:49 [WARN] Sync point out of acceptable range 1954.99 < 1800
13:30:49 [WARN] Sync point out of acceptable range 1985.01 < 1800
13:30:49 [WARN] Sync point out of acceptable range 1834.99 < 1800
13:30:49 [WARN] Sync point out of acceptable range 1934.99 < 1800
13:30:49 [INFO] No movement detected, max gyro angle: 3.7371868872758434. Skipping sync point.
13:30:49 [INFO] Setting offset at 11760.0000: -405.0100 (cost 93997.8837)
13:30:49 [INFO] Setting offset at 18480.0000: -1785.0100 (cost 186195.5933)
13:30:49 [INFO] Setting offset at 21840.0000: 1014.9900 (cost 205376.7073)
13:30:49 [INFO] Setting offset at 25200.0000: -585.0100 (cost 79267.5589)
13:30:49 [INFO] Setting offset at 28560.0000: 834.9900 (cost 69841.6482)
```

panzeromnom, 2022-02-09:
Gyro data:
[WIT5.TXT](https://github.com/gyroflow/gyroflow/files/8033142/WIT5.TXT)

Video File:
https://we.tl/t-I181EJjsRf

panzeromnom, 2022-02-09:
Edit!
Same thing happens when using the AKAZE-Method

Gro2mi, 2022-02-14:
Please include the lens profile otherwise we can't sync succesfully. 

> I hit Autosync with the following parameters selected: 2,0s offset, 10,00s sample size, 10 sample points. 
No, you used following parameters: 0s offset, 2s search size, 10 sample points. Your actual offset seems to be around -5s and 10 sample points are too many. Normally 2-3 sample points are enough.
You're sync is completely off which results in huge and wrong offsets for the sync points and bad stabilization. 
Setting all offsets back to 0 results in displaying the chart correctly. 

[image]

This  seems to be a chart issue in the UI
moving a single syncpoint by ~1ms can result in scattered lines in the chart
[image]
[image]

panzeromnom, 2022-02-14:
My bad.
I used the included "BMPCC 4k Laowa 7,5mm" Preset.
--> That's the setup I`m usin
[Blackmagic_Pocket_Cinema_4K_Laowa_7_5_mm_f2_0_C4k.zip](https://github.com/gyroflow/gyroflow/files/8059320/Blackmagic_Pocket_Cinema_4K_Laowa_7_5_mm_f2_0_C4k.zip)
g

panzeromnom, 2022-02-14:
I did a new test! Please see below:

I set the WIT sensor to 1000Hz sample rate.

It was attached to the camera like this:
[image]

The Camera is set to : 25FPS 1/50 F 2.8

I figured the IMU Orientation could be ZYx.

I first started the sensor to start recording and then the camera.
So i set a rough offset of 2 sek.

As recomended i used max 3 Sync points with 2 seconds.

[image]

After hitting Autosync the graphs disappeared in some areas.

When deleting every Sync point but one, it'll show again.

Videofile:
https://wetransfer.com/downloads/541b8f34491ba5a15909f512bd557ef020220214120749/e959f8861395bb76b55784088169fd4a20220214120824/48081f

Gyrodata:
[WIT8.TXT](https://github.com/gyroflow/gyroflow/files/8060562/WIT8.TXT)

Lens Profile:
[Blackmagic_Pocket_Cinema_4K_Laowa_7_5_mm_f2_0_C4k.zip](https://github.com/gyroflow/gyroflow/files/8060579/Blackmagic_Pocket_Cinema_4K_Laowa_7_5_mm_f2_0_C4k.zip)

Using RC-3

Log Output:

12:08:48 [INFO] Latest version: v1.0.0-rc3, current version: 1.0.0-rc3
12:08:48 [INFO] Lens profiles directory: "\\\\?\\C:\\Users\\Paul\\Documents\\GyroflowRC3\\camera_presets"
12:08:48 [INFO] Loaded 1291 lens profiles in 18.659ms
12:08:48 [INFO] OpenCL Platform: NVIDIA CUDA, Device: NVIDIA Corporation NVIDIA GeForce GTX 1080 Ti
12:08:48 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: create AV_HWDEVICE_TYPE_CUDA
12:08:48 [DEBUG] (1) gyroflow::rendering::ffmpeg_hw: created ok AV_HWDEVICE_TYPE_CUDA
[src\rendering\mod.rs:45] gpu_type = NVIDIA
12:09:32 [DEBUG] (18) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
12:09:32 [DEBUG] (19) MDKPlayer: [in m_player->onEvent] "thread.video" ""
12:09:32 [DEBUG] (20) MDKPlayer: [in m_player->onEvent] "thread.audio" ""
12:09:32 [DEBUG] (20) MDKPlayer: [in m_player->onEvent] "decoder.audio" "FFmpeg"
12:09:32 [DEBUG] (20) MDKPlayer: [in m_player->onEvent] "reader.buffering" ""
12:09:32 [DEBUG] (21) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
12:09:32 [DEBUG] (21) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x25c1ecd6030) QSize(1030, 543)
12:09:32 [DEBUG] (19) MDKPlayer: [in m_player->onEvent] "decoder.video" "AVCodec.D3D11"
12:09:32 [DEBUG] (21) MDKPlayer: [in m_player->onEvent] "render.video" "1st_frame"
12:09:32 [INFO] surface size: 2048x1080
12:09:32 [DEBUG] (21) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::Direct3D11
12:09:32 [DEBUG] (21) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x25c1f0eb800) QSize(2048, 1080)
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 0
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 1
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 2
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 3
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 4
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 5
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 6
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 7
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 8
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 9
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 10
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 11
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 12
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 13
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 14
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 15
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 16
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 17
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 18
12:10:05 [DEBUG] (2) gyroflow::rendering::ffmpeg_hw: config null for 19
12:10:05 [DEBUG] (2) gyroflow::rendering: Selected HW backend AV_HWDEVICE_TYPE_NONE () with format None

12:10:10 [WARN] Sync point out of acceptable range 1985.01 < 1800
12:10:10 [INFO] Setting offset at 2560.0000: -1825.0100 (cost 6189.8435)
12:10:10 [INFO] Setting offset at 7760.0000: -1765.0100 (cost 7017.7978)
12:10:10 [INFO] Setting offset at 12920.0000: -1725.0100 (cost 6015.1480)

AdrianEddy, 2022-02-15:
Ok so the chart issue is fixed in 50cc658c85f3453d6c4a472f3cf471d1bae46240
The reason is that your gyrolog from WitMotion is logging in a very slow rate, it's not really suitable for good stabilization. Is there an option to increase the logging rate in WitMotion?

For your first video, either set IMU orientation to `ZyX` or set Yaw rotation to 180.
For second video the default IMU orientation `ZYx` is correct

You definitely need to increase the logging rate to use this for stabilization

closed 2022-02-15 by AdrianEddy

panzeromnom, 2022-02-16:
> 

Hmm that's kind of strange because the Wit Logger is set to 1000Hz... that's the highest it can go.
And i see that others are getting great results with a logging rate of around 400Hz.

I will chek if the SD Card is the bottleneck.

panzeromnom, 2022-02-16:
:(
Turns out the Sensor can't do more than 200 Hz.
Sorry for bothering...