gyroflow/gyroflow #149: Building on generic linux

skrat, 2022-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

HEAD

### What operating system are you using?

Arch Linux

### What GPU are you using?

intel

### What happened?

I'm trying to get this running and packaged on/for arch linux. I first tried the AppImage, that one fails to run, possibly due to poor wayland support in Qt. I can usually workaround this by setting `QT_QPA_PLATFORM=xcb` but in this case this causes segfault in libxcbcommon.so. Leaving it to Qt to choose the platform, I'm getting this:

```
MDK 0.14.0 (git 378299c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang13.0.1 GCC4.2.1; libc++13000; 15:11:36 Feb  1 2022
Fontconfig warning: "/usr/share/fontconfig/conf.avail/05-reset-dirs-sample.conf", line 6: unknown element "reset-dirs"
20:28:37 [WARN] EGL not available
20:28:37 [INFO] Lens profiles directory: "/tmp/.mount_GyroflLtISZ0/camera_presets"
20:28:37 [INFO] Loaded 1289 lens profiles in 12.801ms
20:28:37 [WARN] QRhiGles2: Failed to create temporary context
20:28:37 [WARN] QRhiGles2: Failed to create context
20:28:37 [WARN] Failed to create RHI (backend 2)
20:28:37 [ERROR] Failed to initialize graphics backend for OpenGL.
fish: Job 1, './Gyroflow-linux64.AppImage' terminated by signal SIGABRT (Abort)
```

Mind that I'm running Vulkan and OpenGL apps on daily basis. This seems to be Qt related but I haven't found a solution yet.

Next up is building from source. I (loosely) followed the steps from README, but I had to comment out the `apt-get` calls from the `ext` script as those are only applicable on debian based distros. I also removed the python/pip based Qt installation, as I'm trying to package this for arch linux and use Qt6 from the system /usr/lib and corresponding header files. Still, it seems to be failing at choosing the wrong Qt version / include files. See here https://gist.github.com/skrat/a912edf303926896d2670c585c7f8c02

I'm no rust expert so I was looking how the linking is done against Qt but couldn't find anything.

AdrianEddy, 2022-02-09:
Which Qt version are you trying to use? Looks like the API changed between versions (note we're using private Qt apis). It was tested with Qt 6.2+

Did you also try the prebuilt .tar.gz package?

skrat, 2022-02-09:
I'm trying to build this against Qt 6 (6.2.3), but I have both Qt 6 and Qt 5 on my system, which I believe is common as there are apps for both. I don't have any Qt 6 apps though. Could you enlighten me how the Qt libs and headers are located and how the CC command args are built? I only found something called qttyped in the deps.

Yes I did try that one too, it seems to work, the app runs. Haven't tried using it yet.

I'm trying to package this for Arch Linux so I need to be able to build this from source and link against system libs. Either that or get the AppImage working.

AdrianEddy, 2022-02-09:
Qt uses qmake in PATH, so `which qmake` should give you the Qt 6.2.3 one to work correctly

skrat, 2022-02-09:
Now we're getting somewhere, so the way this works in this distro, at this moment, is that `qmake` refers to Qt 5 qmake, and `qmake6` refers to Qt 6 qmake. So... another env. var for specifying path to qmake?

AdrianEddy, 2022-02-09:
yes, you can set `QMAKE` environment variable

AdrianEddy, 2022-02-12:
did you manage to compile it successfully? Can this issue be closed? Do you want to add some steps to README that would make the compiling easier for others in the future?

maricn, 2022-02-14:
Sorry for taking over your thread, @skrat,  but hope some of my findings would help you. I come from ArchLinux + swaywm + pipewire (pulse, alsa).

## AppImage

> I'm trying to get this running and packaged on/for arch linux. I first tried the AppImage, that one fails to run, possibly due to poor wayland support in Qt. I can usually workaround this by setting QT_QPA_PLATFORM=xcb but in this case this causes segfault in `libxcbcommon.so`.

I found help for that on https://github.com/gyroflow/gyroflow/issues/154 (repackaging the AppImage without those libs). However that got me through starting the app, but as soon as I load first video in it, it crashes due to:
```
11:48:09 [ERROR] wgpu init error: Any { .. }
ALSA lib dlmisc.c:287:(snd1_dlobj_cache_get) Cannot open shared library /usr/lib/x86_64-linux-gnu/alsa-lib/libasound_module_pcm_pulse.so ((null): /usr/lib/x86_64-linux-gnu/alsa-lib/libasound_module_pcm_pulse.so: cannot open shared object file: No such file or directory)
capi::version: 0.8.2
[1]    15073 segmentation fault (core dumped)  QT_QPA_PLATFORM=xcb ./Gyroflow-linux64_FIXED.AppImage
```

## Building from Source
Then I also went back to build from source, and also hit an error, although a different one. At the last step of the build, running `cargo run --release` produced:
```
   Compiling telemetry-parser v0.1.8 (https://github.com/AdrianEddy/telemetry-parser.git#73a4703a)
error[E0583]: file not found for module `ardupilot`
  --> /home/nikola/.cargo/git/checkouts/telemetry-parser-19821e7ac6d09f51/73a4703/src/lib.rs:10:1
   |
10 | mod ardupilot;
   | ^^^^^^^^^^^^^^
   |
   = help: to create the module `ardupilot`, create file "/home/nikola/.cargo/git/checkouts/telemetry-parser-19821e7ac6d09f51/73a4703/src/ardupilot.rs" or "/home/nikola/.cargo/git/checkouts/telemetry-parser-19821e7ac6d09f51/73a4703/src/ardupilot/mod.rs"

error[E0412]: cannot find type `ArduPilot` in module `ardupilot`
  --> /home/nikola/.cargo/git/checkouts/telemetry-parser-19821e7ac6d09f51/73a4703/src/lib.rs:72:29
   |
72 |     ArduPilot => ardupilot::ArduPilot,
   |                             ^^^^^^^^^ not found in `ardupilot`
   |
help: there is an enum variant `crate::SupportedFormats::ArduPilot`; try using the variant's enum
   |
72 |     ArduPilot => crate::SupportedFormats,
   |                  ~~~~~~~~~~~~~~~~~~~~~~~

Some errors have detailed explanations: E0412, E0583.
For more information about an error, try `rustc --explain E0412`.
error: could not compile `telemetry-parser` due to 2 previous errors
warning: build failed, waiting for other jobs to finish...
error: build failed
```

I wonder have you had issues with `ardupilot` when building and have you made it through with your `QMAKE` envvar?

AdrianEddy, 2022-02-14:
run `cargo update`, this is due to a dependency update yesterday

maricn, 2022-02-14:
@AdrianEddy thank you, that helped!
btws:
1. for building OpenCV with vcpkg, `gperf` needs to be installed... probably to be added to the scripts or docs as a dependency..
2. and also, the gyroflow build proceeds even if OpenCV external dependency build fails.. it only complains at the very end that the version of OpenCV is `0.0.0`..
3. then compiling `qml-video-rs` is problematic and i had to manually download `mdk-sdk` from sourceforge..

after dealing with those, the compilation works and the software can run.. however several other issues pop up.. most notably, loading any file results in:
```
thread '<unnamed>' panicked at 'Platform::default(): Unable to get platform id list after 10 seconds of waiting.', /home/nikola/.cargo/registry/src/github.com-1ecc6299db9ec823/ocl-0.19.3/src/standard/platform.rs:177:55
21:04:38 [ERROR] OpenCL error: Any { .. }
21:04:38 [DEBUG] (13) gyroflow_core::gpu::wgpu: WGPU adapter: AdapterInfo { name: "Mesa Intel(R) UHD Graphics 620 (WHL GT2)", vendor: 32902, device: 0, device_type: IntegratedGpu, backend: Gl }
21:04:38 [DEBUG] (13) MDKPlayer: [in VideoTextureNodePriv::createTexture] QSGRendererInterface::OpenGL
21:04:38 [DEBUG] (13) MDKPlayer: [in MDKPlayer::sync] created texture QSGPlainTexture(0x7fd044c05570) QSize(1280, 720)
thread '<unnamed>' panicked at 'Platform::default(): Unable to get platform id list after 10 seconds of waiting.', /home/nikola/.cargo/registry/src/github.com-1ecc6299db9ec823/ocl-0.19.3/src/standard/platform.rs:177:55
21:04:48 [ERROR] OpenCL error: Any { .. }
21:04:48 [DEBUG] (13) gyroflow_core::gpu::wgpu: WGPU adapter: AdapterInfo { name: "Mesa Intel(R) UHD Graphics 620 (WHL GT2)", vendor: 32902, device: 0, device_type: IntegratedGpu, backend: Gl }
21:04:48 [WARN] Failed to compile shader: 
Source was:
#version 150

struct buf
{
    mat4 matrix;
    float opacity;
};

uniform buf ubuf;

in vec4 vertexCoord;
out vec4 color;
in vec4 vertexColor;
in float _qt_order;

void main()
{
    gl_Position = ubuf.matrix * vertexCoord;
    color = vertexColor * ubuf.opacity;
    gl_Position.z = _qt_order * gl_Position.w;
}

21:04:48 [WARN] Failed to build graphics pipeline state
21:04:48 [WARN] Failed to compile shader: 
Source was:
#version 150

struct buf
{
    mat4 matrix;
    float opacity;
};

uniform buf ubuf;

in vec4 vertexCoord;
out vec4 color;
in vec4 vertexColor;
in float _qt_order;

void main()
{
    gl_Position = ubuf.matrix * vertexCoord;
    color = vertexColor * ubuf.opacity;
    gl_Position.z = _qt_order * gl_Position.w;
}

21:04:48 [WARN] Failed to build graphics pipeline state
21:04:48 [WARN] The Wayland connection broke. Did the Wayland compositor die?
```

@AdrianEddy do you want me to create a separate issue for that?

AlynxZhou, 2022-03-10:
Hi, thank you for this fantastic project, I am a video uploader and also trying to package gyroflow for Arch Linux, and I already get a working PKGBUILD, I put it on AUR: <https://aur.archlinux.org/packages/gyroflow>, however, I have some suggestions for packageing it.

1. The current instructions for building it on Linux will download and build a lot of dependencies, but when packaging it for distributions, we want to use system libraries. While most dependencies can be changed by ENVs, I find that by default it will try to link opencv statically. Currently I use a patch to remove `static=` in `build.rs`, but I hope we can have a compiling option to choose between static and dynamic linking.
2. I grab `OPENCV_LINK_LIBS` from <https://github.com/gyroflow/gyroflow/blob/master/__env-linux.sh#L10>, but on my system I also need to add `opencv_dnn` to `OPENCV_LINK_LIBS` to build it, is this a bug? Or maybe we should use other way to get those value, maybe `pkg-config`?
3. I use system ffmpeg to build this program, but I get a `libffmpeg.so.5` along with `gyroflow`, but I can safely remove it and run `gyroflow`, so why generate it?
4. I also get a `libmdk.so.0`, which is needed to run this program, but I cannot find anything related in official repo, so I cannot find a package provided it.
5. `gyroflow` needs `camera_presets` files, but it only looks for those on current work dir or binary placed dir, so I have to put them into `/opt` instead of put binary to `/usr/bin`, could we have a compiling option to set data dir perfix? So we can then put `gyroflow` into `/usr/bin` and put `camera_presets` to some place like `/usr/share/gyroflow`.
6. I am not familiar with rust, I'm even not familiar with Qt because I am a GTK dev :laughing: , I grab dependencies list from <https://github.com/gyroflow/gyroflow/blob/master/ext/install-deps-linux.sh>, but I am not sure which is runtime dependency, which is build dependency and which is only used to build or download dependency libraries so I can safely ignore them (maybe `p7zip`, `bison`, `zip`, `unzip`...currently I removed them by manually test). Maybe we should have such a list so it will be easier for distribution maintainers to package it.

AlynxZhou, 2022-03-10:
Plus, I also get such output from makepkg:

```
==> WARNING: Package contains reference to $srcdir
opt/gyroflow/gyroflow
```

I use grep and it tells me there are strings contains build dir in binary, maybe we should remove it, binary should not depends on build dir.

I just tested my package with my A7S3's video, it works fine (no runtime error), so I think it's OK to build it with system libraries.

victorlysak, 2024-03-03:
Hi.
If U want to install on Arch as normal software, than the installation is pretty easy.
`yay -S gyroflow-bin`

On my Arch with Plasma 6 it works well only in X11, not in Wayland.
However, I can make it run if I
`export QT_QPA_PLATFORM=xcb gyroflow`

But I haven't tested the performance yet as I spend 98% of the time in X11, as under Wayland (Nvidia) it is not possible for me to do anything norma.
I mean like in edge, brave, chrome screen is flickering, text jumps all over the place, back and forth when I type, and Arduino IDE (1...) is not possible to use at all.

On X11 Gyroflow works very well. Rendering of a 5312x2988 (5k) 50 fps in 150 mbps hevc, goes at around 30 fps on RTX 2060 Super

The env. vars for graphics I also use are:
`VDPAU_DRIVER=nvidia
LIBVA_DRIVER_NAME=nvidia
__GLX_VENDOR_LIBRARY_NAME=nvidia
CUDA_PATH=/opt/cuda
`

The only problem in x11 is that Gyroflow won't remember maximized window state till next time.
Everything else works well.