gyroflow/gyroflow #151: Not able to run exe file

Outboxfpv, 2022-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.00

### What operating system are you using?

Windows 10 64 bit

### What GPU are you using?

Nvedia Quadro FX 4000

### What happened?

When I am running the gyro flow exe file iam getting this error message given below

**clgetkernelargingo could not be located in dynamic link library** #

### Relevant log output

```shell
clgetkernelargingo could not be located in dynamic link library
```

ElvinC, 2022-02-09:
Is that the full copied error log, and are you sure it wasn't `clgetkernelarginfo` instead of `clgetkernelargingo`?

Outboxfpv, 2022-02-19:
> @Outboxfpv <https://github.com/Outboxfpv> is this still an issue in
> latest dev build? https://gyroflow.xyz/devbuild/?autodownload
> Can this issue be closed?

Tanks its solved. I had to install another graphics card driver

closed 2022-02-19 by AdrianEddy