gyroflow/gyroflow #152: Minor UI improvements

avsaase, 2022-02-09:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

I have a couple of suggestions for minor UI improvements:

- When enabling the LPF on the gyro, default to a reasonable value instead of 0.00Hz. I think 60Hz is a good value to start.
- Similarly with the horizon lock amount, it should default to 100% instead of 0%. I also think the extra decimal of precision is not needed here.
- Remove the thousands separator in the output size. This is just a matter of opinion but I don't think it's necessary for values below 10,000.

I tried to do these things myself but I've never worked with QML before and couldn't get the `defautlValue` to do anything.

avsaase, 2022-02-09:
Another suggestion: the spacebar current toggles whichever setting you used last. It would be better if it always starts/stops the video.

relaxgit, 2022-02-12:
Exactly for the spacebar!
also use a little bigger font size. For example when rendering, i barely can read elapsed time, percentage, etc. Too small on my 27" monitor.

relaxgit, 2022-02-12:
would be nice to zoom in/out timeline with the mouse wheel. No need the ctrl key here. Just like in Magix Vegas. Very comfortable.
what about full screen preview playback while doubleclick on the video window?
or scrub while click+move on the preview window?

many thanks!

ElvinC, 2022-02-13:
Most suggestions added with https://github.com/gyroflow/gyroflow/pull/175
UI scaling can be changed under `advanced`

closed 2022-02-13 by ElvinC

relaxgit, 2022-02-13:
very nice, thank you