gyroflow/gyroflow #153: No video playback on Mac after RC2 update.

AdventureFPV, 2022-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0 rc2

### What operating system are you using?

osx monterey 12.0.1

### What GPU are you using?

Radeon Pro Vega 48 8 GB

### What happened?

After the RC2 update I am not able to see the video playback, I can hear the sound but the video does not play. I can use the playhead to skip around and it works. I went back to RC1 and it is working great. Thanks for all of your hard work.

AdrianEddy, 2022-02-09:
Is it black, or it shows a freeze frame?

AdventureFPV, 2022-02-09:
Freeze frame of where ever I place the play head. I can skip around and the
image updates but stays frozen. Thank you so much for all your hard work.

AdrianEddy, 2022-02-09:
Could you test in this one? https://gyroflow.xyz/devbuild/?autodownload

AdventureFPV, 2022-02-09:
I just tested and it did not work. It was strange it started to play for
about 1 sec the first time I hit play. I made a private unlisted video for
you to see the issue. Video: https://youtu.be/q7MXcxct4Vw

AdrianEddy, 2022-02-09:
hmm, weird that part was not changed between rc1 and rc2. 
Could you please go here: https://github.com/gyroflow/gyroflow/actions and download different builds right after `Release v1.0.0-rc1` one (and go up from there) and let me know which one is the first broken? When you click on the action you'll have "Artifacts" link at the bottom to download gyroflow.zip

AdventureFPV, 2022-02-09:
Okay this is where it Broke:
 Fix gyro rotation
<https://github.com/gyroflow/gyroflow/actions/runs/1806653969>Release #163:
Commit 7e2818d
<https://github.com/gyroflow/gyroflow/commit/7e2818d08f0fac92f81f916fa3e983df19e739fd>
pushed
by AdrianEddy <https://github.com/AdrianEddy>
The release before that #141 works fine

AdrianEddy, 2022-02-09:
ok this leads me to believe that it's an issue with external dependency.

Could you please do one more test:
extract both rc1 and rc2, right click and "Show package contents", then go to `Contents` -> `Frameworks` and copy `mdk.framework` from rc1 and replace it in rc2.
So that rc2 uses the one copied from rc1

You can also try the other way around

Please let me know if that changes anything

AdventureFPV, 2022-02-09:
Yes that fixed it. RC2 is now working on my mac with the mdk.framework from
RC1

AdrianEddy, 2022-02-10:
Fixed in latest dev build https://gyroflow.xyz/devbuild/?autodownload

closed 2022-02-10 by AdrianEddy

AdventureFPV, 2022-02-10:
Truly thank you guys so much for all your hard work. Really an amazing
project. I made a small donation and a video series on my YouTube channel I
tell everyone to donate. Thanks again for the fast response.