gyroflow/gyroflow #154: Appimage build crashes on Pop!_OS 21.10

Wazzaps, 2022-02-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Have you tried the latest build?

- [X] I have tried the latest build

### Gyroflow version

1.0.0-rc2 (gh227)

### What operating system are you using?

Pop!_OS 21.10

### What GPU are you using?

Intel UHD Graphics 620

### What happened?

Running the appimage crashes (see attached logs). I'm running under Wayland but both `QT_QPA_PLATFORM=xcb ./Gyroflow-linux64.AppImage` and `QT_QPA_PLATFORM=wayland ./Gyroflow-linux64.AppImage` crash (though in different ways, see logs)

The .tar.gz version seems to open (haven't tested it with a video yet)

Both the stable and dev versions crash the same way

Thanks for the awesome program btw :)

### Relevant log output

```shell
=== X11 MODE ===
(gdb) r
Starting program: /home/david/Downloads/Gyroflow-linux64.AppImage 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 20657]
process 20652 is executing new program: /tmp/.mount_GyroflpL6saU/AppRun
process 20652 is executing new program: /tmp/.mount_GyroflpL6saU/gyroflow
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
MDK 0.14.0 (git 378299c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang13.0.1 GCC4.2.1; libc++13000; 15:11:36 Feb  1 2022
[New Thread 0x7fffee065640 (LWP 20662)]

Thread 1 "gyroflow" received signal SIGSEGV, Segmentation fault.
0x00007fffee0ea015 in ?? () from /lib/x86_64-linux-gnu/libxkbcommon-x11.so.0
(gdb) bt
#0  0x00007fffee0ea015 in ?? () from /lib/x86_64-linux-gnu/libxkbcommon-x11.so.0
#1  0x00007fffee0ea85e in xkb_x11_keymap_new_from_device () from /lib/x86_64-linux-gnu/libxkbcommon-x11.so.0
#2  0x00007fffee1725d2 in ?? () from /tmp/.mount_GyroflpL6saU//lib/libQt6XcbQpa.so.6
#3  0x00007fffee14f31d in QXcbConnection::QXcbConnection(QXcbNativeInterface*, bool, unsigned int, char const*) () from /tmp/.mount_GyroflpL6saU//lib/libQt6XcbQpa.so.6
#4  0x00007fffee16eeed in QXcbIntegration::QXcbIntegration(QList<QString> const&, int&, char**) () from /tmp/.mount_GyroflpL6saU//lib/libQt6XcbQpa.so.6
#5  0x00007fffee1c53ef in ?? () from /tmp/.mount_GyroflpL6saU/plugins/platforms/libqxcb.so
#6  0x00007ffff4a64c2f in ?? () from /tmp/.mount_GyroflpL6saU//lib/libQt6Gui.so.6
#7  0x00007ffff4a67ac6 in QGuiApplicationPrivate::createPlatformIntegration() () from /tmp/.mount_GyroflpL6saU//lib/libQt6Gui.so.6
#8  0x00007ffff4a688c0 in QGuiApplicationPrivate::createEventDispatcher() () from /tmp/.mount_GyroflpL6saU//lib/libQt6Gui.so.6
#9  0x00007ffff534c846 in QCoreApplicationPrivate::init() () from /tmp/.mount_GyroflpL6saU//lib/libQt6Core.so.6
#10 0x00007ffff4a6a581 in QGuiApplicationPrivate::init() () from /tmp/.mount_GyroflpL6saU//lib/libQt6Gui.so.6
#11 0x00007ffff4a6c494 in QGuiApplication::QGuiApplication(int&, char**, int) () from /tmp/.mount_GyroflpL6saU//lib/libQt6Gui.so.6
#12 0x00005555561b342b in ?? ()
#13 0x000055555571b63f in ?? ()
#14 0x0000555555692083 in ?? ()
#15 0x0000555555739b89 in main ()
(gdb) 

=== WAYLAND MODE ===
(gdb) r
Starting program: /home/david/Downloads/Gyroflow-linux64.AppImage 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Detaching after fork from child process 20730]
process 20725 is executing new program: /tmp/.mount_GyroflrtaPDk/AppRun
process 20725 is executing new program: /tmp/.mount_GyroflrtaPDk/gyroflow
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
MDK 0.14.0 (git 378299c) - Multimedia Development Kit. Copyright (c) 2016-2022 WangBin(QtAV author) <wbsecg1 at gmail.com>
Build for: Linux x86_64; Clang13.0.1 GCC4.2.1; libc++13000; 15:11:36 Feb  1 2022
Gtk-Message: 23:34:40.798: Failed to load module "appmenu-gtk-module"
[New Thread 0x7fffec7f5640 (LWP 20750)]
[New Thread 0x7fffebff4640 (LWP 20751)]
[New Thread 0x7fffeb7b8640 (LWP 20752)]
[New Thread 0x7fffeaf04640 (LWP 20753)]

(gyroflow:20725): GLib-GIO-ERROR **: 23:34:40.904: Settings schema 'org.gnome.settings-daemon.plugins.xsettings' does not contain a key named 'antialiasing'

Thread 1 "gyroflow" received signal SIGTRAP, Trace/breakpoint trap.
0x00007ffff1299d35 in ?? () from /tmp/.mount_GyroflrtaPDk//usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
(gdb) bt
#0  0x00007ffff1299d35 in ?? () from /tmp/.mount_GyroflrtaPDk//usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#1  0x00007ffff129adcd in g_log_default_handler () from /tmp/.mount_GyroflrtaPDk//usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff129b01f in g_logv () from /tmp/.mount_GyroflrtaPDk//usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff129b20f in g_log () from /tmp/.mount_GyroflrtaPDk//usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007fffed6303da in ?? () from /tmp/.mount_GyroflrtaPDk//usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#5  0x00007fffed630a6c in ?? () from /tmp/.mount_GyroflrtaPDk//usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#6  0x00007fffed634afb in g_settings_get_enum () from /tmp/.mount_GyroflrtaPDk//usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#7  0x00007fffed97b9be in ?? () from /tmp/.mount_GyroflrtaPDk//usr/lib/x86_64-linux-gnu/libgdk-3.so.0
#8  0x00007fffed97d278 in ?? () from /tmp/.mount_GyroflrtaPDk//usr/lib/x86_64-linux-gnu/libgdk-3.so.0
#9  0x00007fffed976b2e in ?? () from /tmp/.mount_GyroflrtaPDk//usr/lib/x86_64-linux-gnu/libgdk-3.so.0
#10 0x00007fffed91b9ed in gdk_display_manager_open_display () from /tmp/.mount_GyroflrtaPDk//usr/lib/x86_64-linux-gnu/libgdk-3.so.0
#11 0x00007fffedc1a79a in gtk_init_check () from /tmp/.mount_GyroflrtaPDk//usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#12 0x00007fffedc1a7c9 in gtk_init () from /tmp/.mount_GyroflrtaPDk//usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#13 0x00007fffee0e437b in ?? () from /tmp/.mount_GyroflrtaPDk/plugins/platformthemes/libqgtk3.so
#14 0x00007fffee0db5e0 in ?? () from /tmp/.mount_GyroflrtaPDk/plugins/platformthemes/libqgtk3.so
#15 0x00007ffff4a98390 in QPlatformThemeFactory::create(QString const&, QString const&) () from /tmp/.mount_GyroflrtaPDk//lib/libQt6Gui.so.6
#16 0x00007ffff4a651a8 in ?? () from /tmp/.mount_GyroflrtaPDk//lib/libQt6Gui.so.6
#17 0x00007ffff4a67ac6 in QGuiApplicationPrivate::createPlatformIntegration() () from /tmp/.mount_GyroflrtaPDk//lib/libQt6Gui.so.6
#18 0x00007ffff4a688c0 in QGuiApplicationPrivate::createEventDispatcher() () from /tmp/.mount_GyroflrtaPDk//lib/libQt6Gui.so.6
#19 0x00007ffff534c846 in QCoreApplicationPrivate::init() () from /tmp/.mount_GyroflrtaPDk//lib/libQt6Core.so.6
#20 0x00007ffff4a6a581 in QGuiApplicationPrivate::init() () from /tmp/.mount_GyroflrtaPDk//lib/libQt6Gui.so.6
#21 0x00007ffff4a6c494 in QGuiApplication::QGuiApplication(int&, char**, int) () from /tmp/.mount_GyroflrtaPDk//lib/libQt6Gui.so.6
#22 0x00005555561b342b in ?? ()
#23 0x000055555571b63f in ?? ()
#24 0x0000555555692083 in ?? ()
#25 0x0000555555739b89 in main ()
(gdb)
```

Wazzaps, 2022-02-10:
Seems like the guys over here found a solution: https://github.com/OpenVR-Advanced-Settings/OpenVR-AdvancedSettings/issues/569#issuecomment-890587084

adworacz, 2022-03-03:
I can confirm that following the steps linked by @Wazzaps got the AppImage up and running on my Manjaro/Archlinux install.

Specifically, my steps were:
```sh
./Gyroflow.AppImage --appimage-extract
cd squashfs-root/
find . | grep "libx.*\.so" | xargs rm
cd ..
appimagetool squashfs-root Gyroflow_Fixed.AppImage
./Gyroflow_Fixed.AppImage
```

This fix likely also addresses https://github.com/gyroflow/gyroflow/issues/227. 

So there's something about the `libx*.so` files being included that causes the crash on Manjaro/Archlinux at least.

softwarelifecycle, 2022-09-22:
These steps worked on Ubuntu 22.04.1 LTS!

thanks very much!

AdrianEddy, 2022-11-18:
I believe this is fixed in latest dev build https://gyroflow.xyz/devbuild/?autodownload

closed 2022-11-18 by AdrianEddy

patternseek, 2023-04-29:
This is still happening. I just tried to run 1.5.0 @AdrianEddy and had to do the workaround above to get it to run without segfaulting.

AdrianEddy, 2023-04-29:
Latest dev build is newer than 1.5.0, use the dev build link i posted

patternseek, 2023-04-29:
OK, but the link was posted before 1.5.0 so my point was that it doesn't seem to have been fixed at that point and isn't in the current release build.

It is fixed in the latest dev build though, thanks.