gyroflow/gyroflow #156: Improve `.gyroflow` file

AdrianEddy, 2022-02-09:
- [ ] Include export parameters
- [ ] Include autosync parameters
- [ ] When loading `.gyroflow` file, look for the same video filename in the same path, instead of trying to load video from full hardcoded path (for example when you move video file to another folder)
- [ ] Implement loading this file purely in `core` (parse and load all parameters, useful for plugins). 
- [ ] Maybe implement optional binary packed storage for the gyro data? the files can get pretty big, I imagine using simply `bincode` + base64 would reduce them significantly. We lose portability a bit here, but I think 99% users will use it with our code (either main program or plugins) anyway

pitts-mo, 2022-03-09:
My thoughts:
-I utilize read only access on my source folders so specifying the export path for .gyroflow files is desired.
-It seems desirable to enhance default .gyroflow file naming to differentiate between "with" and basic .gyroflow files.

-I also imagine production environments would desire specifying full paths for all input and output. If a search feature is added to locate moved files then an option to disable this search feature may also be desirable.

-p

AdrianEddy, 2022-04-21:
> * [x]  I utilize read only access on my source folders so specifying the export path for .gyroflow files is desired.

If it can't save the `.gyroflow` file in the video directory, it will ask you for new location

> * [ ]  It seems desirable to enhance default .gyroflow file naming to differentiate between "with" and basic .gyroflow files.

What's the use case? IMO it will create more problems that it solves. You can always look at the file size.

> * [x]  I also imagine production environments would desire specifying full paths for all input and output. If a search feature is added to locate moved files then an option to disable this search feature may also be desirable.

It always tries to load from absolute path first. Only if it's not found it falls back to searching in `.gyroflow` directory

AdrianEddy, 2022-04-21:
Implemented in cec97ec1f750b5e85f4246034827fd0e854d4c7c

closed 2022-04-21 by AdrianEddy