gyroflow/gyroflow #159: Automatic detection of gyro data with an offset - maybe a fix for the Runcam Thumb

tomstom, 2022-02-10:
### Is there an existing feature request for this?

- [X] I have searched the existing requests

### Description

In "normal" situations it should be possible to detect if the gyrodata is shifted by an offset, by calculating the mean over the whole clip.

I ask because the gyrodata of the Runcam Thumb seems to have an offset, therefore it is not possible to stabilze it correctly:

The link to the files: https://www.dropbox.com/s/z3gdv271uhjdlbz/Painless360%20RunCam%20Thumb%20files%20for%20GyroFlow.zip?dl=0
(Posted by Painless360 in the comments of the Video: https://www.youtube.com/watch?v=NEalIp1KBBM)

PS: this is a "copy" of an issue posted in the old repository...

ElvinC, 2022-02-10:
The issue with the Runcam Thumb is not the offset, but has to do with framerate/timing, I'm still working on it with them. Other than that you're talking about auto sync which already corrects for offsets. Is there anything more specific in terms of offset detection? Otherwise this is asking for an improved sync, which there are people working on already.

ElvinC, 2022-02-17:
We've found the cause for the Thumb issues and they're trying to fix it through firmware. Closing this as gyro offset detection is synonymous with auto sync.

closed 2022-02-17 by ElvinC